project-oak/silveroak #40: Banish non-decreasing index vectors (downto style)

satnam6502, 2020-04-01:
Let's adopt a "coding style" for the generated SystemVerilog code where all vectors have decreasing indexes i.e. `logic[7:0]` and never `logic[0:7]`.  This should help to avoid terrible mistakes and errors during the SystemVerilog production process in Haskell, and also conforms to the OpenTitan SystemVerilog style rules.

blaxill, 2020-06-15:
In our higher representations the opposite ordering is possibly more natural:

```Haskell
some_list = 2 :: 3 :: 5 :: 8 :: []

some_list[0] == 2
some_list[1] == 3
some_list[2] == 5

-- Indexing in order of definition
some_list[0:2] == 2 :: 3 :: 5 :: []

-- Indexing in reverse order to definition
some_list[2 : 0] == 2 :: 3 :: 5 :: []
```

Of course this is not directly an issue, they can be different. However I've found some cognitive overhead in converting verilog into our representations. Do you have thoughts on this?

satnam6502, 2020-10-08:
I think we need to re-think whether all indices should be decreasing. Let's tackle that with a different issue.

closed 2020-10-08 by satnam6502