project-oak/silveroak #41: Improve directory structure for Cava

satnam6502, 2020-04-01:
The current directory structure for Cava has grown organically but now I think we should move to something more organized. Currently Cava files are split across three directories: cava (the core system but this also has some examples that serve as light regression tests), cava-examples (which has examples) and a nandgame directory (a Cava version of the nand game done by Ben Laurie in the nand directory). Here's a proposal for a better directory structure. Let's discuss and iterate until we come to something that we can agree upon. Here's the current state of the proposal (which we can edit as we 
discuss).

Key points:

- Move to having everything in a single cava directory, with examples nested underneath and a top level README.md that has a good overall explanation of the Cava project and which is well linked to examples, documentation etc.

- To build the system should just require installation of pre-requisites and then typing `make clean all` or just `make`. Regression tests. could be run by `make test` (and perhaps automatically run as part of `make all`).

- The CI system should make sure to track enough of the code under Cava.

- Right now it is just a few us doing experimental evaluation but at some point we should have a way to incorporating contributed circuits and proofs by others e.g. into a `contrib` directory.

- We need to think of where to put the code for the OpenTitan components and a process for using our generated circuits being incorporated into the OpenTitan flow.

A first cut at directory structure inside the `cava` directory:

- Another directory called `cava` with contains a subdirectory Cava: the directory that contains the core Cava system and "standard" library circuits and proofs. Similar to a cleaned up version of the current cava directory, but with examples etc. removed. Also, this is where the current arrow code will end up (hopefully replacing the monadic version).
- examples: The current `cava-examples` plus any examples we want to pick from the current `cava` directory. The examples will also act as tests and include a process for generating SystemVerilog and testbenches which are run and checked using Verilator.
- contrib: A directory where we can accept contributions. We should seed this with one or two examples and a clear process for how to add circuits and write tests etc.
- A `Makefile` at the top level which builds everything with rules for `all` and `clean`. Each sub-directory should also have a `Makefile` with `all` and `clean` rules for its components.

So from the top-level cava directory:
```bash
$ ls
cava contrib examples Makefile
```

blaxill, 2020-04-01:
> The CI system should make sure to track enough of the code under Cava.

Can you clarify- are you just specifying CI should verify all code under cava, with other code a lower priority?  

> Right now it is just a few us doing experimental evaluation but at some point we should have a way to incorporating contributed circuits and proofs by others e.g. into a contrib directory.

Am I correct thinking this is for specifically circuit related contributions (rather than contributions to core cava which will be merged directly)?

> So from the top-level cava directory:

How about a separate test and example directories, and merging contrib and opentitan into a circuit directory:

~~~
$ ls
cava circuits examples tests Makefile
$ ls circuits
contrib opentitan
~~~

satnam6502, 2020-04-01:
I think CI should verify as much as we possible can, including the example and contrib circuits. That increases the chance of catching breaking changes.
Yes, `contrib` is for circuits that other people might want to make using Cava and have distributed with Cava, as well as proofs about circuits elsewhere in the repository. 
OK: let's separate tests and examples. I've been conflicted about this and at first wanted to conflate them. But it is good to keep examples simple and clear, but tests are often noisy and detailed.
As for a `circuits` directory, do you think we will ever have directories other than `contrib` and `opentitan` in the next year? I worry about a directory containing only two other directories.

blaxill, 2020-04-02:
> As for a circuits directory, do you think we will ever have directories other than contrib and opentitan in the next year? I worry about a directory containing only two other directories.

Thats a good point, perhaps just contrib and opentitan in the base directory.

For existing items not related to cava, are we explicitly deprecating them and removing them at this point?

satnam6502, 2020-04-02:
The other stuff in the hardware-oak repository should stay for now. At some
point we should request a separate cava repository under oak.

satnam6502, 2020-04-02:
It might be better to have two different examples directories each with their own extraction scope:
```console
arrow-examples monad-examples
```

satnam6502, 2020-04-11:
I think we've now got to a much better organization.

closed 2020-04-11 by satnam6502