project-oak/silveroak #50: Investigate selective Coq to Haskell extraction

blaxill, 2020-04-05:
Using `Extraction Library` works but we have little control over how the extraction happens, which means the extracted code is not readable (due to lack of type mapping and also module hierachy), and also causes a lot of files to be generated that we need to make sure are hidden from version control (`.gitignore`).

This might be the best way for us, but another way is to manually extract terms with  giving us more control. For an example [Kami](https://github.com/sifive/Kami) uses constant extract to control exactly what and how terms are extracted:

- https://github.com/sifive/Kami/blob/79f75423d7e023f45d126cb8ae2f0b0eb623286f/Extraction.v
- https://github.com/sifive/Kami/blob/8e77e58f368689d18a796c442c4dd20cff603d94/Tutorial/ExtractEx.v
- https://github.com/sifive/Kami/blob/a32841ebf7021f20a455b2a34e9a55322d621d89/Simulator/CoqSim/RegisterFile.v#L331

blaxill, 2020-10-30:
Extraction is also quite noisy right now, which adds to #305

satnam6502, 2021-01-22:
Let's punt in this until phase 2 since it is not phase 1 blocking.

closed 2021-01-22 by satnam6502