project-oak/silveroak #110: Simplify 'Primitive'

blaxill, 2020-05-15:
The definition of Primitive has become overly restrictive for primitives that wouldn't naturally be written as `input_shape -> output_shape`. We could

- Return 'Primitive' to simpler definition.
- Move any abstractions we want to keep into locations closer to where those abstractions are used.
- It has been noted that the 'shape' provided by primitive is perhaps useful at circuit input/output, but overly restrictive at the primitive definition.

satnam6502, 2020-05-15:
As a point of principle, I would like the`Netlist` type to be a sensible type in Haskell i.e. for it to extract nicely. Currently this is not the case which result in the use of `unsafeCoerce` which is problematic for several reasons.

blaxill, 2020-07-03:
These changes have been made.

closed 2020-07-03 by blaxill