project-oak/silveroak #154: Ensure all files have license header

satnam6502, 2020-06-24:
Please can you ensure all the files you have added contain the license header?
For example, for Coq files:
```
(****************************************************************************)
(* Copyright 2020 The Project Oak Authors                                   *)
(*                                                                          *)
(* Licensed under the Apache License, Version 2.0 (the "License")           *)
(* you may not use this file except in compliance with the License.         *)
(* You may obtain a copy of the License at                                  *)
(*                                                                          *)
(*     http://www.apache.org/licenses/LICENSE-2.0                           *)
(*                                                                          *)
(* Unless required by applicable law or agreed to in writing, software      *)
(* distributed under the License is distributed on an "AS IS" BASIS,        *)
(* WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied. *)
(* See the License for the specific language governing permissions and      *)
(* limitations under the License.                                           *)
(****************************************************************************)
```

blaxill, 2020-07-07:
I've updated most of the files. Currently the arrow aes files have the header missing, which i will update once ive fixed the aes components for the various arrow changes from the past ~week

blaxill, 2020-08-17:
I believe all manually created files now have the license header

closed 2020-08-17 by blaxill