project-oak/silveroak #213: Code/build organization

jadephilipoom, 2020-09-22:
The story for building this project is a little jumbled right now -- lots of different `_CoqProject` files, for instance, and some directories require you to run `make` in other directories first to build dependencies. I think it would be smoother if we had:

- One top-level `_CoqProject` file, no other `_CoqProject`s
- A `make` target that makes it possible to automatically populate the `_CoqProject` file (maybe using `git ls-files` or similar)
- DIfferent `make` targets for different logical subparts of the project (e.g. `make arrow-lib`, `make monad-examples`)

This is not something I'm intending on doing at this exact moment, and it should probably be prioritized behind some other things we're all busy with. But I figured I'd open an issue so that whenever one of us does have a moment, we'll have already discussed it and figured out what makes the most sense!

blaxill, 2020-09-22:
I think it also makes sense to remove remove other inactive sub directories keeping only cava, third_party and nix, to make the focus of the project clearer. It might also be worth thinking about a better structuring of directories under `/cava`, that perhaps can address lack of namespacing better (Coq seems to mostly care about the immediate module name, e.g. `A.C` `B.C` produce name collision warnings and extract poorly?)

jadephilipoom, 2020-09-22:
> I think it also makes sense to remove remove other inactive sub directories keeping only cava, third_party and nix, to make the focus of the project clearer.

:+1: 

One consideration for naming directories under `cava/` is that for COQPATH-based builds it's a requirement that the root-level source directory name matches the library name (so, for instance, the directory `arrow-lib` would need to be renamed to `Arrow` because those files are compiled and imported as `Arrow.Foo`). I think this only becomes a problem if someone using a COQPATH-based build decides they want to have us as a dependency, but it's relatively easy to do and not a bad organizational idea anyway.

satnam6502, 2020-10-09:
Soon I would like to reorganize the repo so that the Cava code is at the top level and everything else is relegated to `experiments` or some such directory. As for the `_CoqProject` level organization, this is why I did things the way they are right now:
* cava/cava/Cava is support to represent the "Cava system" as a self-contained entity containing the core combinators/DSLs and libraries of useful circuits.
* cava/monad-exampkes is supposed to represent some "examples" that use the Cava library so I gave this code its own `_CoqProject`. Likewise for `arrow` and `tests`.

Is this the right way to organize a Coq library and clients of the library? Or should we just one one `_CoqProject` file at the root?

jadephilipoom, 2020-10-12:
I don't think there's really a "right way". What you say does make sense for keeping the core library and examples separate; if they're genuinely being treated as separate libraries, different `_CoqProjects` can make sense. (It's possible to do this examples/core library split with a single `_CoqProject`, too -- see bedrock2's  `noex` target, which builds the core library without examples). I think the main difference between the two techniques is that if using different `_CoqProject`s requires building the entire core library in order to build any example, while using one would only update the bits of the core library that are needed for the particular example. It's not a problem now to build the whole core library, but it might be later if build times go up.

A tangential note that I want to make in this build-system discussion is that I've been hearing some buzz about using [Dune](https://github.com/ocaml/dune) for Coq development -- even Coq itself supports a Dune-based build now. It's probably worth spending a little time to evaluate that option when we reorganize.

jadephilipoom, 2020-10-12:
> Soon I would like to reorganize the repo so that the Cava code is at the top level and everything else is relegated to `experiments` or some such directory.

This especially might help with #248 -- I think it's fair enough to not have the pre-submit check compile the experiments directory. And that would remove a barrier to entry for experimental code; one could merge a work-in-progress that isn't performant yet.

closed 2020-11-09 by satnam6502