project-oak/silveroak #235: Prove that build_netlist preserves semantics

jadephilipoom, 2020-10-07:
Requires #234 

This is a necessary link in the chain for arrow-based AES proofs; in order to bridge the gap between the arrow representation and the netlist representation, we need to prove that the translation between these two layers (done by the `build_netlist` routine) preserves semantics.

jadephilipoom, 2020-10-29:
Updating this a bit based on a better understanding than I had before. More generally than just talking about `build_netlist`, we need to more broadly *figure out how* to discuss the semantics of the netlist layer in proofs, and _then_ prove that `build_netlist` preserves semantics.

satnam6502, 2020-10-30:
What is the combinational semantics of ("A >=> B")[Netlist] in the netlist interpretation?
It is specified by the combinational interpretation of the same structure: ("A >=> B")[Combinational]

closed 2020-10-30 by satnam6502

satnam6502, 2020-10-30:
Another attempt to explain what is the intention here. We could do our verification work for phase 1 having only the combinational and sequential interpretations: we don't need to the netlist interpretation to exist.
* The combinational and sequential interpretations/semantics are like the "the programs a user writes in a language for some compiler" and we want check the meaning of these programs against a specification that we write as a Gallina function. That is akin to trying to verify a C program.
* The netist interpretation corresponds to "the C compiler". Verifying things about the netlist interpretation and its mapping down to SystemVerilog is like verifying the C compiler. I consider this out of scope for phase 1.
I hope this makes things clearer.