project-oak/silveroak #238: Emit literal numerals in systemverilog

blaxill, 2020-10-08:
Currently when building a constant vector e.g. `VecLit [Gnd, Vcc, Gnd]` the emitted systemverilog outputs the value directly:

```
  assign v1 = '{zero,one,zero};
```

We could instead output a literal: 

```
  assign v1 = 3'h2;
```

satnam6502, 2020-10-30:
To help with debugging generated SystemVerilog.

satnam6502, 2020-11-02:
@blaxill please can you confirm you want the literals in hex?
Also, please can you confirm _why_ you want the literals? I think you said for debugging? In that case, is it sufficient to just emit a comment after the assignment which gives the value in binary, decimal and hex?
Thanks.

satnam6502, 2021-02-04:
I think the new scheme that emits hex is good enough for now.

closed 2021-02-04 by satnam6502