project-oak/silveroak #248: Pre-submits take too long

satnam6502, 2020-10-11:
The pre-submit process for PRs takes very long. Can we do anything to speed it up?
I assume some of this is due to the number of git submodules we use?

satnam6502, 2020-10-11:
We could consider going back to the previous situation, where we assume the modules we depend are pre-requisites that have already been installed elsewhere by the user.

blaxill, 2020-10-12:
The majority of time is spent in building our own items. I think the first step here is #213 along with also refactoring proofs into separate files, and then trying to extract more parallelism from our build.

jadephilipoom, 2020-10-12:
Is there a way to maintain our own system snapshot that has all the dependencies installed already?

Looking at the build stats for the most recent PR, I'm seeing:

```
Total time: 12:06
---
0: recover_git_folder 00:32
1: cache_dependencies 01:29
2: build_verilator    01:47
3: build_third_party  03:06
4: run_makefile       06:48
```
Step `4: run_makefile` runs `make install` on the dependencies, after having previously compiled them in step 3. That isn't strictly necessary -- we can use the local build instead of installing them globally on the system, and potentially save some time. I'm not sure how much, though, because I'm not seeing timestamps in the log.

PR #306 (2020-10-29): Run `make` in parallel mode for presubmit
If it works, fixes #248 

I have no idea if this works, but I need to create a PR for it for the build to run.
closed 2020-10-29 by jadephilipoom