project-oak/silveroak #268: Transient CI build errors

satnam6502, 2020-10-14:
This issue tracks transient build errors on CI. Whenever I notice one I'll add a note here.

satnam6502, 2020-10-14:
https://pantheon.corp.google.com/cloud-build/builds/ffad1c6d-3abd-47e7-a41f-4643b8ea4b14?project=oak-ci

```console
Step #0 - "recover_git_folder": Cloning into 'oak-hardware-git'...
Step #0 - "recover_git_folder": POST git-upload-pack (142 bytes)
Step #0 - "recover_git_folder": POST git-upload-pack (445 bytes)
Step #0 - "recover_git_folder": mv: cannot move 'oak-hardware-git/.git' to './.git': Directory not empty
Finished Step #0 - "recover_git_folder"
ERROR
ERROR: build step 0 "nixos/nix:2.3.4" failed: step exited with non-zero status: 1
```

blaxill, 2020-10-14:
The cause is actually the start of the build, GCB is starting to more frequently initialize a git repository:

```
FETCHSOURCE
Initialized empty Git repository in /workspace/.git/
From https://github.com/project-oak/oak-hardware
 * branch            cd8c0b08123b0b318ce6dbeb12622d3971c78ee1 -> FETCH_HEAD
HEAD is now at cd8c0b0 Merge arrow-lib back into Cava
```

Previously it would not 

```
FETCHSOURCE
Fetching storage object: gs://691249393555.cloudbuild-source.googleusercontent.com/b3eb8b460810e9d7f73c1278623fc1779fc7fabc-e30ff79c-2123-43ee-b469-56dbfd84fd4c.tar.gz#1602639191658367
Copying gs://691249393555.cloudbuild-source.googleusercontent.com/b3eb8b460810e9d7f73c1278623fc1779fc7fabc-e30ff79c-2123-43ee-b469-56dbfd84fd4c.tar.gz#1602639191658367...
/ [0 files][    0.0 B/ 33.2 MiB]                                                
/ [1 files][ 33.2 MiB/ 33.2 MiB]                                                
Operation completed over 1 objects/33.2 MiB.  
```

This behaviour is non-deterministic as far as I can tell. I think for now I'll update the build steps to be resilient to either initialisation routine.

blaxill, 2020-10-14:
Rebasing on main since #269 (commit 382a11b on main branch) *should* now work.

satnam6502, 2020-10-15:
Got a build failure for Verilator.
```
Step #2 - "build_verilator": error: unable to download 'https://www.veripool.org/ftp/verilator-4.028.tgz': Couldn't resolve host name (6)
Step #2 - "build_verilator": (use '--show-trace' to show detailed location information)
Finished Step #2 - "build_verilator"
ERROR
ERROR: build step 2 "nixos/nix:2.3.4" failed: step exited with non-zero status: 1
Finished Step #3 - "build_third_party"
```
https://pantheon.corp.google.com/cloud-build/builds/29076f6e-27a9-4d6d-8df2-51c0756d25c4?project=oak-ci

blaxill, 2020-10-26:
Verilator fail seems to have been due to a temporary network failure connecting to veripool.org

I think we are good for now, closing as fixed. Please reopen or create a new issue if you see CI build errors unrelated to your PRs.

satnam6502, 2020-12-04:
I think this is a CI failure I just got: https://pantheon.corp.google.com/cloud-build/builds/c47a3f74-1b71-42ef-a7e1-15e8a1a60e69?project=oak-ci

jadephilipoom, 2021-03-16:
https://pantheon.corp.google.com/cloud-build/builds/d4096351-926c-4b5e-aab4-d284f6fcf824;step=3?project=oak-ci

Failed to build `third_party` with the somewhat mystifying error "no targets" on `coqutil`. Possibly the `git submodule update` hadn't yet created the `coqutil` submodule by the time it tried to build it?

blaxill, 2021-03-16:
Odd, it looks like the submodule checkout should have finished at the line

```Submodule path 'bedrock2/deps/coqutil': checked out 'f0eb54037cde200031536a1b49cb528435abe3c9'```

I noticed after pushing your PR worked- did the last force push contain any changes to make it work?

jadephilipoom, 2021-03-16:
> I noticed after your latest push your PR worked- did the last force push contain any changes to make it work?

Nope; the last push was just a `git commit --amend` to get the same commit with a new hash. I also hit "retry" on the failing build and got a [successful one](https://pantheon.corp.google.com/cloud-build/builds;region=global/c4d57dc7-1e39-4331-9333-e4b0fd2ab075?project=oak-ci).

jadephilipoom, 2021-03-16:
Huh, same error again. https://pantheon.corp.google.com/cloud-build/builds/bf65af98-a62c-493c-9d9a-70e75f18f9ab;step=5?project=oak-ci

This time in `make-all`, but *still* about `coqutil`, which is even more confusing. The PR only changes the tutorial file and the documentation, so I don't think it's related? But two errors in four runs is suspicious.

jadephilipoom, 2021-03-16:
Looks like `coqutil` (and the other subdirectories) are missing from `.PHONY`. Not sure if that's causing the problem, but I'll fix it and we'll see?

https://github.com/project-oak/silveroak/blob/1e654d404031cc76b392cd4bc74aafa3e2c09a74/third_party/Makefile#L17

jadephilipoom, 2021-03-17:
I think this is not a problem with the CI system; I just ran into it locally.

closed 2021-03-17 by jadephilipoom