project-oak/silveroak #325: Enable Module Hierarchy in Arrows

blaxill, 2020-11-02:
The arrows representation is currently does not define or produce a module hierarchy. This has been unblocked by the netlist layer allowing module instantiation. This is e.g. a requirement for generating the AES netlists, which otherwise would be completely flattened (and _really_ huge as result).

satnam6502, 2020-11-02:
Note that the netlist layer would have no problem generating a flattened AES circuit for any of the variants, and it won't stress Verilator or the Xilinx tools either. I also don't expect it to be any issue for the monad/Acorn kernel to produce a flattened netlist. You already generate falttened netlists for with the AES layer so I assume you don't have capacity limits there. Do yo want a flattened netlist to help with debugging at the Verilator level or ILAs on the FPGA? I was happy to add hierarchical modules to the netlis but I'd be interested to know how we are using this feature (i.e. what problem is it solving for us right now?). Hierarchy could be useful for analysis by other external tools, although that remains a speculative advantage right now.

blaxill, 2020-11-02:
> However this is marked as a P1 right now. So is the lack of module hierarchy blocking progress? Specifically, what is the problem that is a blocking issue?

Currently SystemVerilog generation various Arrow AES components completes fine (e.g. sboxes), but trying to generate SystemVerilog for `unrolled_cipher_flat` in `unrolled_opentitan_cipher.v` hangs in the compiled Haskell generator (>30mins, will see if it completes over night). Note that the definition evaluates correctly on values. There are other ways to debug this, but delimiting hierarchies would be helpful for some other structures too.

Happy to put this on the back burner for another solution of netlist generation.

satnam6502, 2020-11-02:
Oh no, hanging for 30 minutes+ does not sound good. That should not be happening! Let's talk.

satnam6502, 2020-11-09:
*Question*: to what extent is this about introducing some form of hierarchy in the Coq/DSL layer which is totally independent of the netlist layer hierarchy?

blaxill, 2020-11-09:
> _Question_: to what extent is this about introducing some form of hierarchy in the Coq/DSL layer which is totally independent of the netlist layer hierarchy?

This is indeed independent of the netlist layer, although a prerequisite for the DSL stack to produce netlist hierarchy. The Kappa expressions previously did not explicitly delimit module boundaries (although informally `RemoveContext` mostly was used to delimt module boundaries). Adding module boundaries to Kappa expressions enables a number of improvements including better `CircuitInterface` handling for Kappa expressions, and more importantly for simple caching when lowering expressions to arrow targets which should address performance issues.

satnam6502, 2020-11-19:
Is this still an issue? How much effort in days would it take to resolve this if it is still an issue?

blaxill, 2021-01-14:
#427

closed 2021-01-14 by blaxill