project-oak/silveroak #326: Clean up Arrow notation

blaxill, 2020-11-02:
Some constructs are a lot more verbose than intended, which makes writing the expressions more difficult than necessary. e.g.

~~~
Definition aes_key_expand
  (sbox_impl: SboxImpl)
  : << Bit (* op_i *)
    , Vector Bit 4 (* round id *)
    , Vector Bit 8 (* rcon input *)
    , Vector (Vector (Vector Bit 8) 4) 8 (* input key *)
    , Unit
    >> ~>
      << Vector Bit 8, Vector (Vector (Vector Bit 8) 4) 8>> :=
  <[\op_i round_id rcon key_i =>
~~~

Typed named arguments would be preferable:

~~~
Definition aes_key_expand
  (sbox_impl: SboxImpl)
  : (Vector Bit 8, Vector (Vector (Vector Bit 8) 4) 8) :=
  <[\(op_i: Bit)
       (round_id: Vector Bit 4)
       (rcon: Vector Bit 8)
       (key_i: Vector (Vector (Vector Bit 8) 4) 8) =>
~~~

jadephilipoom, 2020-11-02:
Try this [Coq manual entry](https://coq.inria.fr/distrib/current/refman/user-extensions/syntax-extensions.html?highlight=notations#notations-with-recursive-patterns-involving-binders). Essentially, if you write a notation like:
```coq
Notation "\ x .. y => z" := Abs (fun x => .. (Abs (fun y => z)) ..)
  (at level 200, x binder, y binder, right associativity).
```
...then it will work on both `\ op_i round_id rcon`, or `\ (op_i : Bit) (round_id : Vector Bit 4) (rcon : Vector Bit 8)`, or `\ op_i (round_id rcon : Vector Bit 4)`, or any of the usual binder/type combinations.

blaxill, 2020-11-02:
Unfortunately the binder ares of type `var T` rather than just `T`

blaxill, 2020-11-09:
Going forward with Jade's suggestion in #335, although writing types requires an extra underscore due to `var` e.g. `\(op_i: _ Bit)`.

blaxill, 2021-01-14:
#427

closed 2021-01-14 by blaxill