project-oak/silveroak #338: Normalize imports automatically

jadephilipoom, 2020-11-10:
See https://github.com/project-oak/oak-hardware/pull/330#issuecomment-721030565

In #330, I went through and fully-qualified every single import in our repo to remove undefined behavior and import-ambiguity warnings. But this is also painful to maintain. We should use existing tools, for instance from [coq-tools](https://github.com/JasonGross/coq-tools), to create a script that can just be run before submitting a PR. We could even add it as a git hook.

satnam6502, 2020-11-16:
GitHub hook will check and complain.
Provide fixup script.

jadephilipoom, 2020-11-25:
Brief update on this one after I spent some time looking into it:
- There's a script in `coq-tools` that minimizes imports, with the option to also fully-qualify them. This is cool because it actually finds imports you're not using and removes them, but it's also fairly slow; I wouldn't want to run it as a githook.
- There's currently a bug that affects some of our Export statements -- I opened https://github.com/JasonGross/coq-tools/issues/55. It's pretty hotfix-able for us, so not a blocking issue.
- Unfortunately, I'm not finding a top-level script that fully-qualifies imports without doing extra things. There are some promising-looking sub-functions; we can almost definitely write our own script to do it, but since this is a P2 I'm prioritizing some other things above doing that for now.

jadephilipoom, 2020-12-01:
Another look at this. Turns out there is in fact a script in coq-tools that fully qualifies imports (absolutize-imports.py, I feel silly for not seeing it -- it was hiding by not being executable like the other top-level scripts). I had to modify it slightly to read library information from `_CoqProject` and to not try to rebuild files. From there I was able to write a script that finds all the `_CoqProject`s in our repo (excluding `investigations` and `third_party`) and absolutizes the imports of files they cover.

However, a few caveats:
- The script uses `.glob` files, so it requires the Coq source to be built before it can qualify the imports. That's not very convenient for a git hook, because it will either ignore files that aren't up-to-date or rebuild the files it needs.
- We discussed having this as a pre-push hook, but if you push a branch you're not currently on or have local modifications then things get tricky; I think we'd have to checkout the branch being pushed in order to run the script and see if anything changed.

I see a few possible solutions to the problems above:
1) We could make it a CI check instead of a git hook; slower feedback but we'd know all the files are built and the right branch is checked out. Unfortunately, it would only run once *all* the files have built, so I forsee a lot of "I waited for the whole build and everything works, but it failed because of one not-fully-qualified import" frustration.
2) We could forget about coq-tools and add a simpler git hook that just analyzes the diff being pushed and blocks you if you add a Require line that doesn't start with the name of some linked library (determined by looking at _CoqProject for the relevant file). We'd have to disallow `From _` Requires (which the coq-tools script already does) because there wouldn't be a good way for us to tell if `From Coq Import Foo` is fully-qualified or not.
3) We could do both, with the idea that the git hook is a quick-and-dirty check that catches most things and the CI is more robust.

Any thoughts on these options, or other ideas?

satnam6502, 2020-12-01:
Rather than having something in a githook perhaps it is fine to manually run the script every now and then. Thank you very much for the great investigation.

jadephilipoom, 2020-12-02:
I think I'll go ahead and submit a PR with the script. For now we can just run it intermittently (and encourage people to run it before submitting PRs), and then we can add it to CI/git hook later if we want.

I'll also write a similar wrapper script that calls minimize-requires (without absolutizing, so we don't run into the bug I mentioned), which we can also just run intermittently. But probably a separate PR for that, just to keep the changes readable.

PR #373 (2020-12-07): Remove unused imports
Resolves #338 

Mixed feelings on this one. It turns out that the minimize-requires script from coq-tools is a bit fragile. In particular, I ran into the following issues:
- The script doesn't handle "transitive" requires -- if you import Bar, and Bar imports Foo, you can access things from Foo as e.g. `Foo.x`. But the minimize-requires script considers itself free to remove the Bar import, causing errors because now `Foo.x` is undefined.
- Following the same style as absolutizing (see #368) was really tricky because it kept trying to rebuild files that were out of sync;  [... cut]
closed 2020-12-07 by jadephilipoom