project-oak/silveroak #434: Consider changing the type of indexAt to be dependent

satnam6502, 2021-01-15:
Currently the type of `indexAt` is:
```coq
  indexAt : forall {t : SignalType} {sz isz: nat},
            signal (Vec t sz) ->     (* A vector of n elements of type signal t *)
            signal (Vec Bit isz) ->  (* A bit-vector index of size isz bits *)
            signal t;                (* The indexed value of type signal t *)
```
However, we should consider changing the type to:
```coq
  indexAt : forall {t : SignalType} {isz: nat},
            signal (Vec t (2^isz)) -> (* A vector of n elements of type signal t *)
            signal (Vec Bit isz) ->   (* A bit-vector index of size isz bits *)
            signal t;                 (* The indexed value of type signal t *)
```
i.e. the index value can never be out of range.

CC: @jadephilipoom

jadephilipoom, 2021-01-18:
This might cause problems with proving that vector lengths correspond; my usual heuristics would suggest putting this requirement as a precondition in proofs instead. I think it boils down to this question: do we expect to ever be doing proofs parameterized over vector length instead of parameterized over the log of vector length?

jadephilipoom, 2021-01-18:
The heuristic I'm applying here is "move as much complexity as possible away from the typechecker and into interactive proofs", because it's easier to make an interactive proof go through than to fight with the typechecker.

Thinking about this a little more, one option could be to make `SignalType` only support vectors whose sizes are a power of two; that is, `Vec Bit n` would be a `2^n`-bit vector. In that case, I think all our proofs would end up with that log-value in context, and it wouldn't cause problems. However, the downside is that then inductive proofs would be inductive on the log-value, which might make for trickier vector proofs because dividing-in-half is not the nicest way to do induction on vectors.

satnam6502, 2021-01-18:
I think constraining all vectors to have a length that is a power of 2 would be too restrictive. Perhaps having a `IndexForWord` declaration of some kind would be helpful?

jadephilipoom, 2021-01-18:
Here's a suggestion: why not use the `nth_default` from `VectorUtils` for `indexAt`? I've found its semantics the easiest to reason about for proofs.

satnam6502, 2021-01-19:
Yes, I think `nth_default` ~ which is the current behaviour nayway ~ is the best way forward.
We need to adhere to SystemVerilog coding guidelines for OpenTitan which say when the indexed array is shorter than the range of the index and the array must be extended with a default value. That's exactly what the semantics we have in Coq do and that's what the generated SystemVerilog should do to be compliant with the OpenTitan SystemVerilog coding guidelines.

closed 2021-01-19 by satnam6502