project-oak/silveroak #437: Improve directory / module structure

satnam6502, 2021-01-18:
The top level import needed to start designing circuits:
```
silveroak/cava/Cava/Cava.v
```
This imports `Core` and parts of `Cava.Lib` that are available by default for circuit designs e.g. `Cava.Lib.Prelude` and `Cava.Lib.Combinators`.
```coq
Require Import Cava.Cava.
```
The core system:
```
silveroak/cava/Cava/Core
silveroak/cava/Core/CavaClass.v
silveroak/cava/Core/Netlistv
```
Library definition i.e. those that don't need special access to internals and are just defined in terms of `Core` Cave. Some of these are re-exported by `Cava.v`
```
silveroak/cava/Cava/Lib
silveroak/cava/Cava/Lib/Combinators.v
silveroak/cava/Cava/Lib/Adders.v
silveroak/cava/Cava/Lib/Subtractors.v
silveroak/cava/Cava/Lib/Registers.v
silveroak/cava/Cava/Lib/Multiplexors.v
```

satnam6502, 2021-01-19:
I'd like to make progress on this soon but it's likely to be a disruptive change to inflight PRS.
Right now we have one big PR in flight: #431
Let's wait for it to merge and then I shall re-organize the directory/module structure.

jadephilipoom, 2021-03-08:
I took another look at this issue now that phase 1 is complete. How does this structure sound?

```
Cava/
    Core/
        CavaClass.v
        Netlist.v
        Signal.v
        Circuit.v
        Core.v (exports all of the above)
    NetlistGeneration/
        NetlistGeneration.v
    Semantics/
        Combinational.v
        CombinationalProperties.v (only the proofs that do not relate to combinators)
        Simulation.v
    Lib/
        Adders.v (= combined circuit definitions from UnsignedAdders.v + FullAdder.v + XilinxAdder.v)
        AddersProperties.v (= UnsignedAddersProofs.v + proofs from FullAdder.v and XilinxAdder.v)
        Combinators.v
        CombinatorsProperties.v (= much of the current CombinationalProperties.v)
        Multiplexers.v
        MultiplexersProperties.v
        Prelude.v (= CavaPrelude.v)
        Vec.v
        VecProperties.v
        Lib.v (exports all of the above definitions -- not *Properties.v)
        LibProperties.v (exports all of the above proofs -- everything *Properties.v)
    Util/
        BitArithmetic.v
        Identity.v
        List.v (= ListUtils.v)
        Nat.v (= NatUtils.v)
        Tactics.v
        Vector.v (= VectorUtils.v)
    Cava.v (exports everything needed to write + test circuits)
    CavaProperties.v (exports extra files needed for proofs)
```

satnam6502, 2021-03-08:
Oveall looks great as is. What would get re-exported from `Cava.v`? Perhaps:

* All of `Core`
* `NetlistGeneration`
* `Lib.Combinators`, `Vec.v` --- or export everything in `Lib.v ?
* `Util.BitArithmetic`? 

Perhaps I am just a bit paranoid about this but I do worry about names of modules that are also names in Coq elsewhere e.g. `List.v`, `Nat.v` and `Vector.v`. Again, I don't feel strongly enough to demand different names, but perhaps this is totally fine idiomatic Coq?

jadephilipoom, 2021-03-09:
Actually, I have acquired the heuristic that using the same naming as the standard library, specifically for these kind of "util" proofs, is desired; that way, when someone is working on a proof, they don't have to remember whether a lemma was defined in our repo or in the standard library. If I'm working on a proof with both lists and vectors in scope, for instance, I might have multiple versions of lemmas with the same name e.g. `List.map_id_ext` and `Vector.id_ext`, and need to prefix those names. It's convenient to not have to look up whether I need to write `List.map_id_ext` or `ListUtils.map_id_ext`, and the former is slightly more readable. Ultimately, for proofs at the same level of abstraction as the standard library, it's nice to maintain the illusion that they're the same as the standard library lemmas.

I don't have a strong preference here, but there seems to be a norm in this direction for Coq projects; looking at our dependencies, both `coq-ext-lib` and `coqutil` have called their list-related files `List.v`, so I think it's at least not a glaring error.

jadephilipoom, 2021-03-09:
> Oveall looks great as is. What would get re-exported from `Cava.v`? Perhaps:
> 
> * All of `Core`
> * `NetlistGeneration`
> * `Lib.Combinators`, `Vec.v` --- or export everything in `Lib.v ?
> * `Util.BitArithmetic`?

I agree with all these -- maybe we should actually separate BitArithmetic out into definitions and proofs separately. I lean towards just exporting everything in `Lib.v` for now, since we don't have anything too specialized. If we get to the point where `Lib` is too large, we can think about segmenting more, but I don't think we're at that point yet.

satnam6502, 2021-03-09:
> I agree with all these -- maybe we should actually separate BitArithmetic out into definitions and proofs separately. I lean towards just exporting everything in `Lib.v` for now, since we don't have [...]

Separating `BitArithmetic` sounds like a good call.

PR #636 (2021-03-10): Remove references to "acorn" name
Resolves #437 

Last piece of the big reorganization. Since `acorn` is no longer a separate concept from `cava`, I renamed everything with "acorn" in its name. Name changes were:
- `acorn-examples` (directory) => `examples`
- `AcornExamples` (module name of `acorn-examples`) => `Examples`
- `aes/Acorn` (directory) => `aes/Impl`
- `AcornAes` (module name of `aes/Acorn`) => `AesImpl`

I chose "Impl" for the `aes/Acorn` directory because its sibling directory is `Spec`, and that seemed like a natural contrast.
closed 2021-03-10 by jadephilipoom