project-oak/silveroak #457: Rename `peel` and `unpeel` as `unpackVec` and `packVec`

satnam6502, 2021-01-22:
I find the names `peel` and `unpeel` confusing. Let's think of a better name e.g. `unpackVec` which takes a `SignalType` vector and returns a `Vector.t` and `packVec` which takes a `Vector.t` and returns a `SignalType` vector.

Another question: currently the unpacked type is `Vector.t`. Should it be `list`?

jadephilipoom, 2021-01-22:
I think it may well be worth exploring unpacking `Vector.t` to `list` at some point -- but it would break a lot of current proofs, so perhaps best to try it after our phase 1 deadline.

blaxill, 2021-01-22:
Some additional name ideas:
- bundle / unbundle /  bundleVec / unbundleVec 
- distribute / collect 

I'm happy with any, pack/unpack would get a vote from me just for simplicity.

satnam6502, 2021-01-22:
I like `pack` and `unpack` although I worry about name clashes because they are quite common identifier names. For phase 1 leaving them as `peel` and `unpeel` might be the best plan.

satnam6502, 2021-01-22:
Let's close this for now and revisit in phase 2.

closed 2021-01-22 by satnam6502