project-oak/silveroak #472: Reconsider git ls-files | grep "\.v$" >> _CoqProject

satnam6502, 2021-01-22:
I am not sure if we want to have the Coq source files discovered in this way:
```
git ls-files | grep "\.v$" >> _CoqProject
```
I often trip up when added a new file which has not had `git add` applied to it.
When we have some stability perhaps it is not too bad just to explicitly list the source files in `_CoqProject`.
CC: @jadephilipoom

jadephilipoom, 2021-01-25:
I'm fine with making this change; `git ls-files` is just a default I carried with me from a previous project. I'd prefer not to list everything manually in `_CoqProject`, though, because the reason I made the change is that we had multiple instances of a file accidentally not getting added and therefore not being checked by CI. However, listing all `.v` files whether or not they're known to git seems like a reasonable solution.

satnam6502, 2021-01-26:
An alternative is to just add every `.v` file that can be found:
```bash
find . -name "*.v" >> _CoqProject
```

PR #548 (2021-02-18): _CoqProject generation w/ find.
Closes #472 

This change:
- declares '.v' files as prerequisites for generating _CoqProject
  (avoids running rule, and thereby touching _CoqProject, if no .v
  files are changed, for faster incremental builds).
- uses 'find' over 'git ls-files'. Making code git-agnostic,
  specifically when creating a new file, don't need to `git add` in
  order to build it.
- adds assumption that file names don't have spaces

Future considerations:
- avoid 'xilinx' subdir excludes, for consistent style
- consider factoring out file lookup logic into a define or include
  whic [... cut]
closed 2021-02-18 by jadephilipoom