project-oak/silveroak #488: Create scripts to build OpenTitan with SilverOak replacements

blaxill, 2021-01-27:


blaxill, 2021-01-28:
OpenTitan notes. After tools are installed:
- `fusesoc --cores-root . run --flag=fileset_top --target=sim --setup --build lowrisc:systems:top_earlgrey_verilator`
- `./meson_init.sh `
- `ninja -C build-out all` (this builds aes_test.c, so re-run if disabling mask)
- `build/lowrisc_systems_top_earlgrey_verilator_0.1/sim-verilator/Vtop_earlgrey_verilator   --meminit=rom,build-bin/sw/device/boot_rom/boot_rom_sim_verilator.elf --meminit=flash,build-bin/sw/device/tests/aes_test_sim_verilator.elf   +UARTDPI_LOG_uart0=-` AES test execution with verilator and UART out redirected to console

blaxill, 2021-02-10:
This has been completed for the combinational components, with `aes_cipher_core` remaining.

jadephilipoom, 2021-08-23:
Looking at the discussion in https://github.com/project-oak/silveroak/pull/464, it appears this work was completed; I'm closing the issue.

closed 2021-08-23 by jadephilipoom