project-oak/silveroak #505: git clean -Xf is a big productivity loss for me

satnam6502, 2021-02-02:
I keep getting stung by the `git clean -Xfd` command in our Makefiles. I've lost about half an hours work writing a tcl script to drive FPGA implementation for the AES block because I forgot to add this tcl script to the `.gitignore`.

I think we should reconsider a scheme that is so brutal about deleting files and that requires us to explicitly whitelist files before typing `make` otherwise they get deleted.

blaxill, 2021-02-03:
I'd like to have generated files in specific directories, that way we could nuke only those directories and know everything else should be kept. I think for the Coq extraction this means having an extraction directory. 

Happy for any other fix in the meantime.

jadephilipoom, 2021-02-03:
We can take care of Coq build artifacts like `.vo` files by using `make -f Makefile.coq clean`. Doing that plus a directory for extraction files sounds like a good solution to me!

jadephilipoom, 2021-08-23:
We now use `make -f Makefile.coq clean`, so I think the problem is fixed; closing this issue.

closed 2021-08-23 by jadephilipoom