project-oak/silveroak #526: Fix naming and organization in silveroak-opentitan/aes/Acorn

jadephilipoom, 2021-02-09:
Currently the naming and organization for the Acorn AES circuits is a bit chaotic. In particular:

- Circuit names are `add_round_key`, `sub_bytes`, `aes_mix_columns`, and `aes_shift_rows`; we should decide whether we want `aes_` prefix or not
- The circuit and tests are in separate files for `mix_columns` (`MixColumnsCircuit.v` and `MixColumnsTests.v`) but not for any other subcircuit
- The file containing the top-level cipher is called `CipherRound.v` but actually contains the full cipher definition
- We're missing a test for the inverse direction of `shift_rows`
- We only use the `AES256.from_flat` and `AES256.to_flat` conversions in some of the tests but ad-hoc transformations in others

For most of these, I think it matters more that we have a convention than what the convention is. However, this is how I would resolve them if I were to do it right now:
- Add `aes_` prefix to `sub_bytes` and `add_round_key`
- Create new `*Tests.v` files for all the subcircuits and move tests there + rename circuit files to `*Circuit.v`
- Rename `CipherRound.v` to `Cipher.v`
- Add inverse test for `shift_rows`
- Use  `AES256.from_flat` and `AES256.to_flat` conversions everywhere

Any thoughts/opinions?

jadephilipoom, 2021-02-10:
#527 fixed the issues with tests, leaving the `aes_` prefix and `CipherRound.v` naming still to be resolved.

blaxill, 2021-02-10:
#532

closed 2021-02-10 by blaxill