project-oak/silveroak #552: muxPair/pairSel potentially creates Pairs in Vectors

blaxill, 2021-02-18:
Pairs in Vectors become a problem for our SystemVerilog generation and shouldn't be emitted in our netlists. Unfortunately since muxPair/pairSel now put their arguments into a vector to be indexed, nested Pairs will create Pairs in Vectors when passed to muxPair.

satnam6502, 2021-02-18:
And just to be clear, we _could_ define SystemVerilog record types to represent each instance of pairs, but I'd rather we did not and avoided all issues related to nominal vs. structural typing. Also, although we emit SystemVerilog, I'd like to keep open the door to also emitting Verilog, especially for processing by Lutsig, which means no records.

jadephilipoom, 2021-03-01:
Resolved by #595, where pairs were removed entirely.

closed 2021-03-01 by jadephilipoom