project-oak/silveroak #565: Port tests to new sequential setup

jadephilipoom, 2021-02-19:
See #561 -- code under `tests` should be ported to use the new `Circuit` inductive so that the old `seqType`-based `Cava` instance can be eventually removed.

smore-lore, 2021-02-24:
After https://github.com/project-oak/silveroak/pull/589

I think we only have these left:
acorn-examples/IncrDecr.v:24:Require Import Cava.Acorn.CombinationalProperties.
cava/Cava/Acorn/SequentialProperties.v:31:Require Import Cava.Acorn.CombinationalProperties.

So I think we can also remove/rename these as part of this task.

jadephilipoom, 2021-02-25:
You're correct, once this is finished we can do some cleanup. That cleanup, including the removing/renaming, is already tracked in a separate issue, though: #567

PR #592 (2021-02-25): Port remaining tests and examples
Resolves #565 

There might still be some small cleanup to do, but all the proofs and large examples are now ported to `Circuit`. Once #589 is merged, we're clear to start #567.
closed 2021-02-25 by jadephilipoom