project-oak/silveroak #568: Add reset and enable signals to Circuit constructors

jadephilipoom, 2021-02-19:
Following the principle of having the most generic base constructors possible, we should add reset and enable signals to the base `Loop` and `Delay` constructors of circuit (and rename them to something like `LoopRE`) and then define the loop with no extra signals in terms of these.

I believe I know how to do reset values, but I'm less clear on adding enables. A few questions:
- What should the netlist for a loop-with-enable look like? Looking at the existing netlist generation code, it seems like it's the same as a normal loop, but with a delay-with-enable instead of a delay.  What does this mean in terms of semantics? Does it mean the value held in the delay doesn't get updated if the enable is false?
- The most natural way I'd define a loop without enables is to fix the enable bit to `one`. Is it okay if every single loop we generate has enables, just with some of them wired to `Vcc`? Will SystemVerilog optimize this away to a loop/delay without an enable?
- In the `interp` function, I'll need to somehow (without knowing what the `Cava` instance is) construct the monadic expression for the loop so that it performs a selection on the state. For this I think I need a generic `mux2` in `CavaClass` -- otherwise I have no way to do the selection in a generic way. Would it be acceptable to add `mux2`?

satnam6502, 2021-02-19:
A point of pedantry: we are not wiring up reset signal: we are providing an _initial value_ for the register after the reset has been applied. The wiring of the reset signal occurs outside the context of Cava descriptions and our semantics depend on reset being asserted just once at the very beginning of time and never again.

It's perfectly fine to tie the enable input to false/`Vcc` : the correct optimizations will occur and we will still get efficient circuis.

Yes, let's have a generic `mux2`, I was on the verge of adding it back anyway. We need a way of making a choice between two generic things based on a boolean value, and that's why `mux2` was in the class in the first place.

blaxill, 2021-02-19:
- A clock enable disables the clock when de-asserted, this effectively disables a circuit stopping (registered) parts of the circuit from changing. In terms of semantics, the state would no longer update and (optionally but likely preferred due to consistency) any delays within the subcircuit should also have updates suspended while the clock enable is de-asserted.   

- I think it should be optimised away, but we could also just expose the primitive with an optional enable signal.

- I'm not against adding `mux2` but I believe this can be done with `indexAt` such as the existing `muxPair` : https://github.com/project-oak/silveroak/blob/2fb9a39b849e16d969f0ce073b9ac317cac265ac/cava/Cava/Acorn/CavaPrelude.v#L45-L49

satnam6502, 2021-02-19:
> * A clock enable disables the clock when de-asserted, this effectively disables a circuit stopping (registered) parts of the circuit from changing. In terms of semantics, the state would no longer [...]

Indeed, and I am happy either way, making it a primitive or defining it in terms of indexing a two-element vector.
Either way, if we wanted to do selection between two _composite_ collection of signals (e.g. `signal a * signal a`) then we would have to write a specialized function for each case (shape) so the muxing is broken down to apply to just `SignalType` values or we'd have to find some scheme (e.g. using typeclasses) that provided a more generic `mux2` that worked over any nesting of tulpes of values and generated the appropriate muxing.

jadephilipoom, 2021-02-19:
Thanks both! I think I'm clear on what to do now.

PR #572 (2021-02-19): Add enables to Circuit
Resolves #568 

Now we have four options for both `Loop` and `Delay`:
- `LoopInitCE` and `DelayInitCE`, with both enables and reset values, which are the base constructors in terms of which the rest are defined
- `LoopInit` and `DelayInit`, which have reset values but not enables
- `LoopCE` and `DelayCE`, which have enables but default reset values
- `Loop` and `Delay`, which have no enables and default reset values
closed 2021-02-19 by jadephilipoom