project-oak/silveroak #575: Revert indexAt and indexConst to being circuits

satnam6502, 2021-02-23:
They used to be circuits which made resource usage clear. I changed them to expressions but they've ended up being used in a way that does not make resource usage clear, which is shame because that is definitely something want to have a firm control of. Revert to previous definitions. Hopefully minimal impact on proof.
```coq
(* Dynamic indexing *)
  indexAt : forall {t : SignalType} {sz isz: nat},
            signal (Vec t sz) ->              (* A vector of n elements of type signal t *)
            signal (Vec Bit isz) ->           (* A bit-vector index of size isz bits *)
            cava (signal t);                  (* The indexed value of type signal t *)
  (* Static indexing *)
  indexConst : forall {t: SignalType} {sz: nat},
               signal (Vec t sz) ->           (* A vector of n elements of type signal t *)
               nat ->                         (* Static index *)
               cava (signal t);               (* The indexed bit of type signal bit *)
```

CC: @blaxill @jadephilipoom

blaxill, 2021-02-23:
Additionally `slice`, `unsignedAdd`, `unsignedMult`, `greaterThanOrEqual` are not assigned wires and could result in duplication? (Also pairing but paring is to be removed)

satnam6502, 2021-02-23:
> Additionally `slice`, `unsignedAdd`, `unsignedMult`, `greaterThanOrEqual` are not assigned wires and could result in duplication? (Also pairing but paring is to be removed)

Yes, indeed!

jadephilipoom, 2021-03-01:
Resolved by #596

closed 2021-03-01 by jadephilipoom