project-oak/silveroak #591: Remove the need for unIdent

jadephilipoom, 2021-02-25:
Possibly controversial, but I think it's a bit awkward to have to write `unIdent` everywhere. The reason is that `coq-ext-lib` defines the identity monad with:
```coq
Inductive ident A := mkIdent { unIdent : A }.

Global Instance Monad_ident : Monad ident :=
{ ret  := fun _ v => mkIdent v
; bind := fun _ _ c f => f (unIdent c)
}.
```
However, there's no need to define it with a record type this way. I would have written:
```coq
Definition ident A := A.

Global Instance Monad_ident : Monad ident :=
{ ret  := fun _ v => v
; bind := fun _ _ c f => f c
}.
```
This would simplify how we talk about the correctness of combinational circuits; `unIdent (foo x y) = foo_spec` would become just `foo x y = foo_spec`.

We also wouldn't lose any helpful lemmas from `coq-ext-lib`, because there just aren't any -- the only file that seems to contain identity monad lemmas is completely commented out. I think it's just in general a slightly neglected piece of code, and we wouldn't lose anything by defining a nicer one ourselves.

blaxill, 2021-02-25:
I think this is reasonable. The reason it has a trivial constructor is likely due to Haskell roots. It needs a datatype in Haskell because class instances are unique to a datatype (and are also how they are dispatched). 

The only question I have is does it make typeclass instance resolution more difficult? I can imagine this simpler definition will collide with other possible instances or make currently non-ambiguous cases ambiguous.

jadephilipoom, 2021-02-25:
You're right that this version is somewhat more ambiguous, but Coq doesn't require unique solutions for typeclass inference, and I think the ambiguity might be okay here.

I did some quick experiments with a minimal setup. I copied the `Monad` typeclass and notations from `coq-ext-lib` and defined the identity monad as a `Global Instance`, then created a section with a different monad as a context variable to mimic the kind of setup we often have. Then I tried `Check (x <- ret 5 ;; ret (x + 1))` to see which monad instance got used. Here, both definitions of `ident/Monad_ident` behaved identically; the local monad instance from the section took precedence.

A difference emerged, though, when the monad instance was ill-formed. I tried `Check (x <- ret 5 ;; (x + 1))` (missing the last `ret`) and got an error with the `coq-ext-lib` definition and no error with my proposal (the `ident` monad was picked up instead because it could in fact typecheck). With typeclass constraints on the expression, `Check (x <- ret 5 ;; (x + 1) : m nat)`, both versions of `ident/Monad_ident` produced an error.

Personally, I'd be willing to accept this ambiguity since I don't expect we would be importing the identity monad instance in very many places where it wouldn't be used, and it would save us a lot of boilerplate. But it is a tradeoff -- if anyone has objections, I'm happy to leave it as-is.

satnam6502, 2021-03-03:
I agree the `unIden` is annoying. I'd vote for giving this a go with an alternative identity monad that does not wrap.

PR #627 (2021-03-05): Remove unIdent
Resolves #591 

Cleans up our lemma statements and single-step simulations of combinational circuits by using our own identity monad that does not have a wrapper record type.
closed 2021-03-05 by jadephilipoom