project-oak/silveroak #634: Clean up examples and tests

jadephilipoom, 2021-03-10:
As we try to make the codebase more presentable, it would be nice to have a solid suite of example circuits/proofs for curious readers to look at. Currently, we have examples scattered between `acorn-examples` and `tests`, with various degrees of presentability. I suggest:
- Selecting the most presentable of these, adding extra explanatory comments, and moving them to an `examples` directory, with definitions and proofs in the same files for easier browsing
- Moving anything less presentable to the `tests` directory
- Adding a suggestion to look at `examples` from the README and the tutorial once it's complete

satnam6502, 2021-03-10:
Yes. I/we should make a wee plan about what we have, what we need and then create tasks for clean thing up etc.

jadephilipoom, 2021-08-23:
No longer relevant now that we've moved to redevelop the core library in cava2; closing the issue.

closed 2021-08-23 by jadephilipoom