project-oak/silveroak #640: Polish contents of Util

jadephilipoom, 2021-03-11:
Similar to #639: now that we're exporting `Util/`, it would be nice to clean it up a bit. This issue tracks those efforts.

- [ ] Remove experimental code and unused definitions (unused lemmas can stay)
- [ ] Write tests for all tactics
- [ ] Organize files more logically
 
A bit more on the last point: many files in `Util` (looking at you, `Vector.v`) contain proofs and code interleaved chaotically. Any consistent structure is a good structure, but my first suggestion for `Util` files would be the style in `List.v`, with a separate section for each definition and its proofs, or just the proofs if it's from the standard library. Example snippet illustrating the pattern:

https://github.com/project-oak/silveroak/blob/b0dc9fd8f33cd41c983c11574b11a3a64ddd153e/cava/Cava/Util/List.v#L164-L227

blaxill, 2021-09-21:
At this point we have `cava2` succeeding `cava` but depending on it for `Cava/Util`.  AFAIK `Cava/Util` has no dependencies and could be made into a standalone library.

jadephilipoom, 2021-09-27:
> At this point we have `cava2` succeeding `cava` but depending on it for `Cava/Util`. AFAIK `Cava/Util` has no dependencies and could be made into a standalone library.

There's already an issue for this: #914