project-oak/silveroak #644: Create separate repo for Github Pages

jadephilipoom, 2021-03-16:
Merging `main` into `gh-pages` can cause issues (see #642). @blaxill suggests making a new repo just for our documentation that has silveroak as a submodule, so we can simply update the submodule when needed. This would streamline the updating process for docs and simplify the main repo by removing extra infrastructure needed only for docs. Thoughts?

jadephilipoom, 2021-03-16:
Looks like we would have to create a new repo under `project-oak` for this. Slightly hesitant about it because there aren't many repos there, and I don't have permissions to create new ones (presumably for a reason).

After some searching around for other ways to set up this kind of workflow, I have two ideas:
1. Change Github Pages look for its source on `main`, and do away with the `gh-pages` branch entirely. We'd have to run CI for documentation changes, but that seems like less of a pain than the issues we're currently having, and less complicated than any other setup.
2. Remove non-website files from `gh-pages` and remove all website-related files from `main`. Making changes to `.md` or other documentation that doesn't depend on `main` would just be done on the `gh-pages` branch. To edit Alectryon-generated interactive pages, we would make changes to the source files based on `main`, test them locally and get them merged to `main` (without the generated HTML). We could then update the live website version by generating the HTML locally on `main` and then copying the result to a locally checked-out version of `gh-pages`, from which we could then make a PR to `origin/gh-pages`.

satnam6502, 2021-03-16:
How does the Oak runtime team manage their GitHub site without a separate repo? @tiziano88  ?

satnam6502, 2021-03-16:
I am hesitant about having a separate documentation repo.

satnam6502, 2021-03-16:
> Looks like we would have to create a new repo under `project-oak` for this. Slightly hesitant about it because there aren't many repos there, and I don't have permissions to create new ones (presumably for a reason).
> 
> After some searching around for other ways to set up this kind of workflow, I have two ideas:
> 
> 1. Change Github Pages to look for its source on `main`, and do away with the `gh-pages` branch entirely. We'd have to run CI for documentation changes, but that seems like less of a pain than the issues we're currently having, and less complicated than any other setup.
> 2. Remove non-website files from `gh-pages` and remove all website-related files from `main`. Making changes to `.md` or other documentation that doesn't depend on `main` would just be done on the `gh-pages` branch. To edit Alectryon-generated interactive pages, we would make changes to the Coq source files based on `main`, test them locally and get them merged to `main` (without the generated HTML). We could then update the live website version by generating the HTML locally on `main` and then copying the result to a locally checked-out version of `gh-pages`, from which we could then make a PR to `origin/gh-pages`.

Right now I'd go for option 1 and suffer the pain of CI, perhaps motivating us to see if we can cache CI builds (more)?

blaxill, 2021-03-16:
It sounds like we need more intelligent CI, so I willl take a look at that- e.g. not doing a full CI run for changes to `doc/`

blaxill, 2021-03-16:
I think we could add `main` as a submodule to `gh-pages`, same repository (different branches) without a new separate repository- but only as a suggestion to make things simpler- don't bother if its more difficult!

satnam6502, 2021-03-16:
> It sounds like we need more intelligent CI, so I willl take a look at that- e.g. not doing a full CI run for changes to `doc/`

Yeah, I think this might be the best approach.

jadephilipoom, 2021-03-16:
Sounds like we do option 1 for now, and deal with the CI wait time until we have a way to avoid triggering it on documentation changes?

In that case, I'll close this issue and make a PR to add the current `docs` content of `gh-pages` to `main`, then change the Pages setup to point to `main`.

satnam6502, 2021-03-16:
> Sounds like we do option 1 for now, and deal with the CI wait time until we have a way to avoid triggering it on documentation changes?
> 
> In that case, I'll close this issue and make a PR to add the current `docs` content of `gh-pages` to `main`, then change the Pages setup to point to `main`.

That's my vote but happy to be outvoted if you two prefer a different option.

closed 2021-03-16 by blaxill

tiziano88, 2021-03-16:
> How does the Oak runtime team manage their GitHub site without a separate repo? @tiziano88 ?
> Perhaps they don't run into the generated HTML issue.
> Perhaps worth looking deeper into GitHub actions?

basically we just have a github action that writes out to the `gh-pages` branch on our own repo, and that gets automatically deployed as a site:

https://github.com/project-oak/oak/blob/7b2fbde00dc9790c413726d1c357fc4fa7be4d9d/.github/workflows/build_rust_docs.yaml#L54-L67

jadephilipoom, 2021-03-17:
> basically we just have a github action that writes out to the `gh-pages` branch on our own repo, and that gets automatically deployed as a site:

This seems like a good setup!