project-oak/silveroak #647: Direct inclusion vs blackbox differs in simulation

blaxill, 2021-03-16:
Currently in the `aes_cipher_core` formulation we include the sub components as blackboxes:

https://github.com/project-oak/silveroak/blob/f8fc0677e19a60d0e7a36e6d6a75fc73605087d3/silveroak-opentitan/aes/Impl/CipherControlNetlist.v#L31-L38

Note that although these could call any Verilog module, they are setup to use our generated files. Changing to direct instantiation works for most of the sub modules but changing `aes_mix_columns` to directly be included causes `aes_cipher_core` to fail tests. Blackbox instantiation and direct inclusion should be semantically equivalent. It could be that `aes_mix_columns` has an issue but this seems to be the same bug seen in #623

satnam6502, 2021-03-16:
Noted! Perhaps a possible bug with black box instantiation.

blaxill, 2021-03-17:
Just to update; I'm still investigating, but it looks clearly an issue with the direct inclusion whereas blackbox is working as expected. Here is the blackbox version of mix_columns which is signalling as expected: 

[image]

Here is the direct inclusion version, clearly showing buggy signals:
[image]
Note the odd periodic transitions before the input even transitions. Perhaps something to do with registering vectors.

blaxill, 2021-03-17:
Perhaps coincidental, but those erroneous transitions are 8 cycles which is the size of the inner most vector (a byte) in the input/output.

blaxill, 2021-03-17:
I've instrumented aes_mix_columns under both instantiations and observed _correct_ signalling in both cases- e.g. although only changing how `aes_mix_columns` is included will trigger the bug, the signals inside aes_mix_columns are correct in both cases.

blaxill, 2021-03-18:
The designs should be able to pass LEC. If they do then this is likely an issue with Verilator.

closed 2021-03-18 by blaxill

satnam6502, 2021-03-18:
For the record.
```
No LSB modules are available.
No LSB modules are available.
2021/03/18 12:56:23 System is not a supported distribution
2021/03/18 12:56:23 An error occurred. We don't recognize OS
2021/03/18 12:56:23 WARNING This OS does not appear to be a Cadence supported Linux configuration.
2021/03/18 12:56:23 For more info, please run CheckSysConf in <cdsRoot/tools.lnx86/bin/checkSysConf <productId>
// LEC version 20.20-p100 (17-Nov-2020)
// Check out Conformal_Ultra 20.2 license
// Check out Conformal_Asic 20.2 license
// Command: read design -golden -root aes_cipher_core -sv12 debug_blackbox/*.sv
// Parsing file debug_blackbox/aes_cipher_control.sv ...
// Parsing file debug_blackbox/aes_cipher_core.sv ...
// Parsing file debug_blackbox/aes_key_expand.sv ...
// Parsing file debug_blackbox/aes_sbox_lut.sv ...
// Parsing file debug_blackbox/aes_pkg.sv ...
// Parsing file debug_blackbox/aes_mix_columns.sv ...
// Parsing file debug_blackbox/aes_pkg.sv ...
// Parsing file debug_blackbox/aes_sbox_lut.sv ...
// Parsing file debug_blackbox/aes_shift_rows.sv ...
// Parsing file debug_blackbox/aes_sub_bytes.sv ...
// Golden root module is set to 'aes_cipher_core'
// Warning: (RTL1.3) Variable/signal is assigned by both blocking and non-blocking assignments (occurrence:172)
// Warning: (RTL7.3) Array index in RHS might be out of range (occurrence:40)
// Warning: (RTL14) Signal has input but it has no output (occurrence:260)
// Warning: (RTL14.1) Fanout load of the signal is removed (occurrence:290)
// Warning: (IGN3.2) Duplicate modules/entities are detected. Subsequent modules/entities are ignored (occurrence:1)
// Warning: (HRC3.10a) An input port is declared, but it is not completely used in the module (occurrence:128)
// Warning: (HRC3.16) A wire is declared, but not used in the module (occurrence:3)
// Note: Read VERILOG design successfully
CPU time     : 1.05    seconds
Elapse time  : 2       seconds
Memory usage : 214.96  M bytes
// Command: read design -revised -root aes_cipher_core -sv12 debug_direct/*.sv
// Parsing file debug_direct/aes_cipher_control.sv ...
// Parsing file debug_direct/aes_cipher_core.sv ...
// Parsing file debug_direct/aes_key_expand.sv ...
// Parsing file debug_direct/aes_sbox_lut.sv ...
// Parsing file debug_direct/aes_pkg.sv ...
// Parsing file debug_direct/aes_mix_columns.sv ...
// Parsing file debug_direct/aes_pkg.sv ...
// Parsing file debug_direct/aes_sbox_lut.sv ...
// Parsing file debug_direct/aes_shift_rows.sv ...
// Parsing file debug_direct/aes_sub_bytes.sv ...
// Revised root module is set to 'aes_cipher_core'
// Warning: (RTL1.3) Variable/signal is assigned by both blocking and non-blocking assignments (occurrence:172)
// Warning: (RTL7.3) Array index in RHS might be out of range (occurrence:40)
// Warning: (RTL14) Signal has input but it has no output (occurrence:260)
// Warning: (RTL14.1) Fanout load of the signal is removed (occurrence:290)
// Warning: (IGN3.2) Duplicate modules/entities are detected. Subsequent modules/entities are ignored (occurrence:1)
// Warning: (HRC3.10a) An input port is declared, but it is not completely used in the module (occurrence:128)
// Warning: (HRC3.16) A wire is declared, but not used in the module (occurrence:3)
// Note: Read VERILOG design successfully
CPU time     : 3.00    seconds
Elapse time  : 4       seconds
Memory usage : 381.41  M bytes
// Command: report design data
============================================
Summary           Golden         Revised
============================================
Design-modules        25               6
Library-cells      10442           10442
============================================
Primitives      Golden              Revised
============================================
INPUT              524                  524
OUTPUT             134                  134
--------------------------------------------
AND               5324                 5324
[... 176 lines cut ...]
                     Compare Result       Golden            Revised
--------------------------------------------------------------------------------
Root module name                          aes_cipher_core   aes_cipher_core
 
Primary inputs                               524                524
   Mapped                                    524                524
 
Undriven key points                            0                 29
   Unmapped                                    0                 29
      Extra                                    0                 24
      Unreachable                              0                  5
 
Primary outputs                              134                134
   Mapped                                    134                134
      Equivalent                134
 
State key points                             535                535
   Mapped                                    535                535
      Equivalent                535
================================================================================
// Command: usage
CPU time     : 4096.58 seconds
Memory usage : 647.01  M bytes
// Command: exit -force
```