project-oak/silveroak #654: Make it easier to build in selected sub-directories

satnam6502, 2021-03-17:
I wanted to just build in the `examples` directory but it no longer has a `Makefile`. So I built one directory above but it tries to make a lot of other things and does some unnecessary recompilation. Is it possible to structure things such that each sub-directory has a stand-alone `Makefile` so you can always just do `make` in a sub-directory? Then the top-level `Makefie` can just invoke the sub-`Makefile`s?

```
satnam-macbookpro3:silveroak satnam$ make examples
/Library/Developer/CommandLineTools/usr/bin/make -C third_party
git submodule update --init --recursive
/Library/Developer/CommandLineTools/usr/bin/make build-coqutil
/Library/Developer/CommandLineTools/usr/bin/make -C bedrock2/deps/coqutil all
Generating Makefile.coq.all
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq.all
make[5]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make build-bedrock2
/Library/Developer/CommandLineTools/usr/bin/make -C bedrock2/deps/coqutil all
Generating Makefile.coq.all
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq.all
make[5]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make -C bedrock2 bedrock2_noex
/Library/Developer/CommandLineTools/usr/bin/make -C /Users/satnam/silveroak/third_party/bedrock2/deps/coqutil
Generating Makefile.coq.all
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq.all
make[6]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make -C /Users/satnam/silveroak/third_party/bedrock2/bedrock2 noex
Generating Makefile.coq.noex
rm -f .coqdeps.d
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq.noex
make[6]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make build-extlib
/Library/Developer/CommandLineTools/usr/bin/make -C coq-ext-lib theories
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq
make[5]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make build-record-update
/Library/Developer/CommandLineTools/usr/bin/make -C coq-record-update
make[5]: Nothing to be done for `real-all'.
/Library/Developer/CommandLineTools/usr/bin/make -C cava
coq_makefile -f _CoqProject -o Makefile.coq
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq
make[3]: Nothing to be done for `real-all'.
./FixupHaskell
mv Ascii2.hs Ascii.hs
mv ByteVector2.hs ByteVector.hs
runhaskell Setup.hs configure --user
Configuring Cava2HDL-0.1.0.0...
runhaskell Setup.hs build
Preprocessing library for Cava2HDL-0.1.0.0..
Building library for Cava2HDL-0.1.0.0..
[55 of 62] Compiling Ascii            ( Ascii.hs, dist/build/Ascii.o, dist/build/Ascii.dyn_o )
[57 of 62] Compiling ByteVector       ( ByteVector.hs, dist/build/ByteVector.o, dist/build/ByteVector.dyn_o )
runhaskell Setup.hs install
Installing library in /Users/satnam/.cabal/lib/x86_64-osx-ghc-8.10.3/Cava2HDL-0.1.0.0-A0nwmqFM9TwLGofj2vO0AK
Registering library for Cava2HDL-0.1.0.0..
/Library/Developer/CommandLineTools/usr/bin/make -C examples
coq_makefile -f _CoqProject -o Makefile.coq
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq
COQDEP VFILES
COQC FullAdderExample.v
COQC IncrDecr.v
COQC NandGate.v
```

jadephilipoom, 2021-03-17:
> I wanted to just build in the `examples` directory but it no longer has a `Makefile`. So I built one directory above but it tries to make a lot of other things and does some unnecessary [...]

I think the system is already set up exactly as you describe. I just checked main and `examples` does have a `Makefile` : https://github.com/project-oak/silveroak/blob/main/examples/Makefile

Did something fail when you changed into `examples` and ran `make`? What error did you get?

blaxill, 2021-03-17:
Moving to a build system such as Dune (which seems favoured amongst recent Coq projects) would also simplify usage and maintenance of this.

jadephilipoom, 2021-08-23:
Closing this issue unless anyone has objections; it's been inactive for a long time, and I think the system is already structured in the way described.

closed 2021-08-23 by jadephilipoom