project-oak/silveroak #657: Implement an efficient constant equality for vectors

satnam6502, 2021-03-17:
Define a combinational circuit parameterized on the size of a vector `n` which takes as input an n-bit vector and returns `true` if the binary value of the vector represents a given compile time `nat` constant `v`, `false` otherwise. This circuit should have an efficient implementation which we hope to use as a component for a decoder circuit as part of another issue.

```coq
Section WithCava.
  Context {signal} `{Cava signal}.

  (* A top-level multiplier circuit that can be compiled to a top-level
    SystemVerilog circuit. *)
  Definition vecConstEq {n v: nat}
                        (a: signal (Vec Bit n)) :
                        cava (signalBit).
     (* Define! *)

End WithCava.
```

You should add your example the Cava's `Lib` directory in `VecConsteq.v`.

**Hint**. Convert the nat `v` into a bit-vector, then perform pair-wise comparison, and then use an AND-tree.
We will want help from @blaxill @jadephilipoom @smore-lore and @cipher1024 

Steps, submitting a PR for review after each step:

- [ ] First define any Gallina function that gives the correct result. Test your function with a few `Example` or `Goal` clauses.
- [ ] Iterate description until you have something that works by comparing bits pairwise and then AND-ing them together (folding the ANDs or using a tree). Check with some tests.
- [ ] Define a circuit interface and then extract both the circuit and a testbench. Add the invocation of the Verilator test-bench for your circuit and make sure the test passes.

satnam6502, 2021-03-17:
@blaxill @jadephilipoom Does this circuit already exist in some form as part of the AES core?

jadephilipoom, 2021-03-17:
> @blaxill @jadephilipoom Does this circuit already exist in some form as part of the AES core?

We already have `eqb`, which works for all signals and does indeed compare with an and-tree (`all`):
https://github.com/project-oak/silveroak/blob/b4be19338faf54e4689c15a3e7bfac5960def10d/cava/Cava/Lib/CavaPrelude.v#L45-L53

If I specialize to a `Vec Bit n` and simplify `eqb` and `all` (`Eval cbv [eqb all] in (fun n => eqb (t:=Vec Bit n))`), I get:
```coq
fun (n : nat) (x y : combType (Vec Bit n)) =>
       eq_results <- Vec.map2 (fun '(a, b) => xnor2 (a, b)) (x, y);; match n with
                                                                     | 0 => ret one
                                                                     | S _ => tree and2 eq_results
                                                                     end
```
Is this the circuit needed for the task?

satnam6502, 2021-03-17:
Well, that's a _dynamic_ equality check rather than checking against a compile time constant?
Of course, this could be used and specialized for make a bit-vector comparison for a constant value.
Let's try that first, and see if synthesis etc. will generate a good circuit.

satnam6502, 2021-03-17:
Another thing to note is that the fully dynamic `eqb` works for any Cava signal type, but `vecConstEq`. We could try to define a `constEq` circuit which takes a compile time constant of the appropriate argument (via `combType`) but I thought as a first starter exercise it might be better to do the more specialized `vecConstEq`. It can be generalized later.

closed 2021-03-19 by atondwal