project-oak/silveroak #663: Move SystemVerilog generation into Coq

blaxill, 2021-03-18:
We have discussed performing all SystemVerilog generation in Coq. Now might be a good time to help with the elusive bug in #647

satnam6502, 2021-03-18:
Indeed, this is a good aspiration and keen to do this at some point, perhaps aligned with generating output that is impedance matched with connecting with LutSig from Chalmers.

If you really think #647 might be due to the Haskell code for netlist generation I could look at that but I suspect if this an error on our part it is more likely to be in the Coq code than the Haskell code.

jadephilipoom, 2021-08-23:
Closing this issue; now that we're moving to cava2, there's no need to port the old netlist generation (although we probably want to implement the new netlist generation in Coq from the start).

closed 2021-08-23 by jadephilipoom