project-oak/silveroak #673: Update coq-tools and try out new features

jadephilipoom, 2021-03-19:
See:
https://github.com/project-oak/silveroak/pull/368#discussion_r597288464
https://github.com/JasonGross/coq-tools/issues/55
#373
#368

Currently we have some infrastructure using `coq-tools` to clean up imports: `minimize-requires` targets that try to get rid of unused imports, and `tools/absolutize_requires.sh` that absolutizes imports. There have been some updates to `coq-tools` since we last set this up; we should update `coq-tools` and see if we can make that infrastructure nicer and potentially even use it in CI. I find I spend a fair amount of time trying to figure out which imports are unused and prune them, so it would be cool to just run `make minimize-requires` and have it taken care of.

blaxill, 2021-03-19:
`make minimize-requires` would definitely be nice. What are you thinking for CI? Require minimised imports before acceptance?

jadephilipoom, 2021-03-19:
> Require minimised imports before acceptance?

Yes -- maybe also absolutized and in alphabetical order. Essentially, I would want to have a `make` target that cleans up the imports, and a CI check that makes sure you've run it (i.e. that running it is a no-op). I think it would be too annoying to have a CI check for imports if the fix was anything other than a one-line command, but if we can manage the one-line command it would be a nice way to keep things organized.