project-oak/silveroak #676: Remove or redefine slice

jadephilipoom, 2021-03-19:
`Cava` contains a field for a `slice` primitive on vectors, which is currently unused:
https://github.com/project-oak/silveroak/blob/064dea72a806a0b032926615737cf692524c53e9/cava/Cava/Core/CavaClass.v#L67-L70

It requires a proof as an argument, which is dangerous for computation performance.

1. Do we need this primitive? I'm generally in favor of minimizing the size of `Cava` where we can, to reduce our trusted interface. If we don't use it and there's no clear reason to keep it, I'd vote for removing it.
2. If we do need it, I'd suggest changing the type signature to `forall {t: SignalType} {startAt len sz: nat}, signal (Vec t (startAt + len + sz) -> cava (signal (Vec t len))` so we get the same guarantee without the proof. We can use `Vec.resize_default` to get a size in this form.

blaxill, 2021-03-19:
I believe its isomorphic to slicing an unpacked vector and so can be removed.

satnam6502, 2021-03-19:
OK! Happy to give it the chop!

PR #677 (2021-03-19): Remove vector slice from Cava class
Resolves #676
closed 2021-03-19 by jadephilipoom