project-oak/silveroak #679: Rename defaultSignal to zero?

jadephilipoom, 2021-03-22:
Just another thought I'm having while tutorial-writing. I think we originally added `defaultSignal` to handle absurd cases where we don't care what's returned. But in some places (for instance, loops or delays with the default initial state) we now consider it okay to rely on the fact that the `defaultSignal` is the "zeroed out" version of the signal. I think a rename might make this clearer.

blaxill, 2021-03-22:
Perhaps we should split the use cases into two definitions - 'absurdSignal' and 'zeroSignal', perhaps that would help if we wanted to prove that 'absurdSignal' is never actually present in a design or similar. 

We could also probably hide the defaults in a lot of places by making a specialised/wrapper for 'resize_default' that works with vectors of signals, and just calls `resize_default default_signal`.

jadephilipoom, 2021-03-22:
> Perhaps we should split the use cases into two definitions - 'absurdSignal' and 'zeroSignal', perhaps that would help if we wanted to prove that 'absurdSignal' is never actually present in a design or similar.

I think that if you use a `defaultSignal` for a certain case, and you prove that the circuit behaves correctly, then whether the case is not supposed to happen or whether your spec is satisfied by zero in that case is sort of an internal detail to the proof. Do you have specific cases in mind where it would be beneficial to expose the difference?

> We could also probably hide the defaults in a lot of places by making a specialised/wrapper for 'resize_default' that works with vectors of signals, and just calls `resize_default default_signal`.

Already done! `Vec.resize_default` uses `defaultSignal` as the default:
https://github.com/project-oak/silveroak/blob/e83fe025e5186c59ea6e31256f4c2f8f86cc5505/cava/Cava/Lib/Vec.v#L144-L147

satnam6502, 2021-03-22:
The concept of a default signal is an important one for the generated SystemVerilog because for each type we generate there is a default value it has in the SystemVerilog semantics and it is very important our definition of default value matches what SystemVerilog does (which at the bit-level corresponds to "all zeros").

blaxill, 2021-03-23:
> I think that if you use a defaultSignal for a certain case, and you prove that the circuit behaves correctly, then whether the case is not supposed to happen or whether your spec is satisfied by [...]

I don't have a concrete use case and so we shouldn't worry too much, but it may be useful to prove non-functional properties - if an implementation unintentionally depends on an 'absurdSignal' it could result in additional unintended resource usage.

closed 2021-08-23 by jadephilipoom