project-oak/silveroak #690: Consider upgrading Haskell compiler to 9.0.1

satnam6502, 2021-03-23:
https://www.haskell.org/ghc/download_ghc_9_0_1.html#macosx_x86_64

satnam6502, 2021-03-23:
Will require work e.g.
```
[ 7 of 62] Compiling Functor          ( Functor.hs, dist/build/Functor.o, dist/build/Functor.dyn_o )

Functor.hs:19:16: error:
    Not in scope: ‘GHC.Base.unsafeCoerce#’
    Perhaps you meant ‘GHC.Base.unsafeChr’ (imported from GHC.Base)
    Module ‘GHC.Base’ does not export ‘unsafeCoerce#’.
   |
19 | unsafeCoerce = GHC.Base.unsafeCoerce#
   |                ^^^^^^^^^^^^^^^^^^^^^^
make: *** [haskell] Error 1
```
@blaxill

atondwal, 2021-04-02:
NB this requires patching the coq haskell extraction plugin, since it still depends on ghc 8, even in the coq master branch (should be a 1-2 line patch, though)