project-oak/silveroak #704: Rework decoder to not use packV and unpackV

satnam6502, 2021-03-25:
The current decoder is defined:
```coq
  Definition map2VC {n a b c}
    (f: a -> signal b -> cava (signal c))
    (v1: Vector.t a n)
    : signal (Vec b n)
    -> cava (signal (Vec c n))
    :=  unpackV
    >=> Vector.map2 (fun f v => f v) (Vector.map f v1)
    >=> mapT_vector id
    >=> packV.

  (* A decoder from binary to one-hot. Both are big endian *)
  Definition decoder {n : nat}
    : signal (Vec Bit n) -> cava (signal (Vec Bit (2^n)))
    := forkN (2^n) >=> map2VC (vecConstEq n) (Vector.vseq 0 (2^n)).

```
which is mainly my fault. It uses `packV` and `unpackV` but really I feel these two operators should only get used in our `Lib/Vec` library to define our overloaded vector operations. So let's rework this. Here are some suggestions:

* Use just a regular `map` (not `map2`).
* Rework `vecConstEq` so it takes a pair, so it can be used as an argument to `map`.
* Define a combinator in `Combinators.v` that "injects a first element: `injFst {a b : SignalType} (x : signal a) (y : signal b) : cava (signal a * signal b)`.
* Now you should be able to write `forkN (2^n) >=> injFst (vseq 0 (2^n)) >=> map vecConstEq` to define the decoder.

CC: @jadephilipoom

atondwal, 2021-03-25:
I don't think we can quite do that: the key reason we had to write `mapVC` is because `vecConstEq` takes one static and one dynamic argument. So, we can't use `Vec.map` because the first argument (the constant to compare to) can't be made into a `signal`, and we can't use `Vector.map` without `unpackV`ing, because the second argument is a wire.

satnam6502, 2021-03-25:
Ah right! I had forgotten about that. 
So here is another proposal. Do away with `vecConstEq` and use something based on `eqb` instead which takes tupled arguments?

jadephilipoom, 2021-03-29:
> @blaxill @jadephilipoom I guess we don't have a `zipWith`?

We do; I think you're looking for `Vec.map2`:
https://github.com/project-oak/silveroak/blob/765519a9e8f2ad308440c2df7d06ba026ec9310c/cava/Cava/Lib/Vec.v#L187-L190

`Vec.map_literal` also will be useful here; it lets you create a signal by mapping a function over a Coq vector that doesn't necessarily have a valid signal type (e.g. a vector of `nat`s).

If I understand the function correctly, I think it would work to simply write:
```coq
Vec.map_literal (fun i => eqb input (nat_to_bitvec i)) (Vector.vseq 0 (2^n)
```

satnam6502, 2021-03-29:
@atondwal Ignore me and so what @jadephilipoom says.

atondwal, 2021-03-29:
So, I'll revert `decoder` to last week's version?

https://github.com/project-oak/silveroak/blob/d0f688dcc2d5b131f952e408eb34fc8cd1cd2126/cava/Cava/Lib/Decoder.v#L25-L29

And I'll rewrite `encoder` to use `map_literal` as well.

satnam6502, 2021-03-29:
Yes, thank you. I am going into a cave to think about this.

commit 6d91b3a0 (2021-03-29): Rewrite encoder/decoder not to use packV (#708)

fixes #704
closed 2021-03-29 by atondwal