project-oak/silveroak #710: Github Actions CI failed where GCB CI succeeded

jadephilipoom, 2021-03-30:
Just wanted to flag this discrepancy (it's not blocking anything since the Actions CI is not required to merge). These two builds are on the same commit https://github.com/project-oak/silveroak/commit/be14f19527e665f42c3e95d2d0f2ff8f09ce3eaf:

https://github.com/project-oak/silveroak/runs/2225830089
https://pantheon.corp.google.com/cloud-build/builds;region=global/38a58a75-d502-4d38-8ffe-4f63b8b0ef4a?project=oak-ci

Confusingly, a change to the tutorial caused `silveroak-opentitan/pinmux` to fail to build, so this seems like a CI issue to me.

CC @blaxill

blaxill, 2021-03-30:
Hmm curl failed, it seems like a transient issue.

~~~
 Cloning into '/home/runner/work/silveroak/silveroak/third_party/opentitan'...
  Error: error: RPC failed; curl 16 Error in the HTTP2 framing layer
  Error: fatal: error reading section header 'acknowledgments'
  Fetched in submodule path 'third_party/bedrock2', but it did not contain f6b8047e8666d73eaa78d61e5478f50accf74c08. Direct fetching of that commit failed.
  Error: The process '/usr/bin/git' failed with exit code 1
~~~

Thanks for reporting though, I will try to investigate a bit

blaxill, 2021-04-06:
Closing for now

closed 2021-04-06 by blaxill