project-oak/silveroak #760: Build break

satnam6502, 2021-04-21:
I've encountered a build break:
```console
./FixupHaskell
mv Ascii2.hs Ascii.hs
mv ByteVector2.hs ByteVector.hs
cabal build
Preprocessing library for Cava2HDL-0.1.0.0..
cabal: can't find source for Number in ., dist/build/autogen,
dist/build/global-autogen
```

Versions:
```console
$ ghc --version
The Glorious Glasgow Haskell Compilation System, version 8.10.3
$ cabal --version
cabal-install version 2.2.0.0
compiled using version 2.2.0.1 of the Cabal library
```
Then I did:
```console
$ git submodule update --init --recursive
```
and got:
```console
$ make
/Library/Developer/CommandLineTools/usr/bin/make -C third_party
/Library/Developer/CommandLineTools/usr/bin/make build-coqutil
/Library/Developer/CommandLineTools/usr/bin/make -C bedrock2/deps/coqutil all
Generating Makefile.coq.all
/Library/Developer/CommandLineTools/usr/bin/make -f Makefile.coq.all
COQDEP VFILES
COQC /Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Ltac2Lib/Msg.v
COQC /Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Tactics/ParamRecords.v
File "/Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Tactics/ParamRecords.v", line 53, characters 8-10:
Error:
Syntax error: [tactic:tac2expr level 6] expected after 'in' (in [tactic:tac2expr]).

```

blaxill, 2021-04-21:
These errors are both from our change to require Coq 8.13- Number has become Numeral, and the updated bedrock requires 8.13

blaxill, 2021-04-21:
We should probably add an early version check to Coq (and other tools) in our Makefile, that would make it clearer when we've increased our lower bound on tools.

satnam6502, 2021-04-21:
Our documentation at https://github.com/satnam6502/silveroak says:
```
The Coq proof assistant version 8.12.0.
```
I think it is very important that I (and others) can robotically and mindlessly just read the documentation and do what it says and have the right thing happen.

blaxill, 2021-04-21:
Ah, that is your fork, the up to date documentation is at https://github.com/project-oak/silveroak and says 

```
The Coq proof assistant version 8.13.0.
```

satnam6502, 2021-04-21:
Ah, I had to hit refresh on my browser! Thanks.

satnam6502, 2021-04-21:
Updated Coq. Did `make clean`. Got:
```console
COQC /Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Tactics/ParamRecords.v
File "/Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Tactics/ParamRecords.v", line 2, characters 0-42:
Error:
File /Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Ltac2Lib/Msg.vo
has bad version number 81200 (expected 81300). It is corrupted or was
compiled with another version of Coq.

make[5]: *** [/Users/satnam/silveroak/third_party/bedrock2/deps/coqutil/src/coqutil/Tactics/ParamRecords.vo] Error 1
make[4]: *** [all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [build-coqutil] Error 2
make[1]: *** [coqutil] Error 2
make: *** [third_party] Error 2
satnam-macbookpro3:silveroak satnam$ pwd
/Users/satnam/silveroak
```

atondwal, 2021-04-21:
You have to `make cleanall`

satnam6502, 2021-04-21:
I don't see that on https://github.com/satnam6502/silveroak

satnam6502, 2021-04-21:
And anyhow, now I get:
```console
cabal build cava-tests
Warning: Unknown/unsupported 'ghc' version detected (Cabal 2.2.0.1 supports
'ghc' version < 8.5): /usr/local/bin/ghc is version 8.10.3
Resolving dependencies...
Configuring cava-tests-0.1.0.0...
cabal: Unknown build target 'cava-tests'.
There is no component 'cava-tests'.

make[1]: *** [TestsSV] Error 1
make: *** [tests] Error 2
```

blaxill, 2021-04-21:
Cabal version 3.0.0.0 or later is required

satnam6502, 2021-04-21:
I did notice that, and I did try:
```console
$ cabal new-install Cabal cabal-install
```
but that did not work!

satnam6502, 2021-04-21:
Specifically:
```console
Configuring cabal-install-3.4.0.0 (exe:cabal)...
Building cabal-install-3.4.0.0 (exe:cabal)...
Generating cabal-install-3.4.0.0 (exe:cabal) documentation...
cabal: symlink-bindir is not defined. Set it in your cabal config file or use
--symlink-bindir=<path>

satnam-macbookpro3:silveroak satnam$ cabal --version
cabal-install version 2.2.0.0
```

blaxill, 2021-04-21:
It looks like cabal doesn't update cleanly? try old install `cabal install cabal-install`

satnam6502, 2021-04-21:
Most of the time I forget why I hate cabal, but then once in a while...

satnam6502, 2021-04-21:
This seems to be stuck:
```console
satnam-macbookpro3:silveroak satnam$ cabal install cabal-install
Resolving dependencies...
Configuring HTTP-4000.3.16...
Configuring cryptohash-sha256-0.11.102.0...
Configuring echo-0.1.4...
Configuring edit-distance-0.2.2.1...
Configuring base16-bytestring-0.1.1.7...
Configuring ed25519-0.0.5.0...
Configuring lukko-0.1.1.3...
Configuring Cabal-3.4.0.0...
Building lukko-0.1.1.3...
Building edit-distance-0.2.2.1...
Building base16-bytestring-0.1.1.7...
Building ed25519-0.0.5.0...
Building echo-0.1.4...
Building cryptohash-sha256-0.11.102.0...
Building HTTP-4000.3.16...
Building Cabal-3.4.0.0...
Installed base16-bytestring-0.1.1.7
Configuring regex-posix-0.96.0.0...
Installed echo-0.1.4
Configuring resolv-0.1.2.0...
Building regex-posix-0.96.0.0...
Installed cryptohash-sha256-0.11.102.0
Installed lukko-0.1.1.3
Building resolv-0.1.2.0...
Installed edit-distance-0.2.2.1
Installed regex-posix-0.96.0.0
Installed ed25519-0.0.5.0
Installed HTTP-4000.3.16
Installed resolv-0.1.2.0
```

atondwal, 2021-04-21:
Hmm, the thing to do might be to update your system version of cabal? Seems
the google repos have 3.0.0.0
```
***@***.***:~$ which cabal
/usr/bin/cabal
***@***.***:~$ cabal --version
cabal-install version 3.0.0.0
compiled using version 3.0.1.0 of the Cabal library
```

satnam6502, 2021-04-21:
I'm on my Mac. It has become unstuck:
```console
Installed Cabal-3.4.0.0
Configuring hackage-security-0.6.0.1...
Building hackage-security-0.6.0.1...
Installed hackage-security-0.6.0.1
Configuring cabal-install-3.4.0.0...
Building cabal-install-3.4.0.0...
```

satnam6502, 2021-04-21:
But still failed to update:
```console
Installed cabal-install-3.4.0.0
Updating documentation index
/Users/satnam/Library/Haskell/share/doc/x86_64-osx-ghc-8.10.3/index.html
satnam-macbookpro3:silveroak satnam$
satnam-macbookpro3:silveroak satnam$
satnam-macbookpro3:silveroak satnam$
satnam-macbookpro3:silveroak satnam$ cabal --version
cabal-install version 2.2.0.0
```
I might just manually install the binary :-(

satnam6502, 2021-04-21:
Manually downloading the cabal binary and running it won't work due to corp policy.

satnam6502, 2021-04-21:
So it looks like I might now be forced to build cabal from source :-(

satnam6502, 2021-04-21:
Let me see if I can do some kind of hack with `.cabal/store/ghc-8.10.3/cbl-nstll-3.4.0.0-e36ff31c/bin/cabal`

satnam6502, 2021-04-21:
Well, that only took just over half a day :-)
```console
           AssignSignal (Wire 0) (IndexConst (LocalVec Bit 4 0) 0)];
         inputs :=
           [{| port_name := "rst"; port_type := Bit |};
           {| port_name := "clk"; port_type := Bit |}];
         outputs := [{| port_name := "o"; port_type := Vec Bit 4 |}]
       |}
     : Module
     = [0%N; 1%N; 1%N; 2%N; 3%N; 5%N; 8%N; 13%N; 21%N; 34%N]
     : list N
     = [0%N; 1%N; 1%N; 2%N; 3%N; 5%N; 8%N; 13%N; 21%N; 34%N]
     : list N
     = (unit * (unit * Vector.t bool ?n) * unit)%type
     : Type
```
Thank you everyone.

closed 2021-04-21 by satnam6502