project-oak/silveroak #784: Cava2HDL does not appear to be installed anymore

satnam6502, 2021-05-14:
I can't load up generated Haskell files or compile them with `ghc --make`.

```
~/silveroak/examples$ ghci AdderTree.hs
GHCi, version 8.10.3: https://www.haskell.org/ghc/  :? for help
Loaded GHCi configuration from /usr/local/google/home/satnam/.ghci
[1 of 1] Compiling AdderTree        ( AdderTree.hs, interpreted )

AdderTree.hs:7:1: error:
    Could not find module ‘Adders’
    Use -v (or `:set -v` in ghci) to see a list of the files searched for.
  |
7 | import qualified Adders
  | ^^^^^^^^^^^^^^^^^^^^^^^

AdderTree.hs:8:1: error:
    Could not find module ‘BinNat’
    Use -v (or `:set -v` in ghci) to see a list of the files searched for.
  |
8 | import qualified BinNat
  | ^^^^^^^^^^^^^^^^^^^^^^^
```

satnam6502, 2021-05-15:
And attempting to install it does not seem to work.
```
$ cabal install Cava2HDL
Warning: The package list for 'hackage.haskell.org' is 483 days old.
Run 'cabal update' to get the latest list of available packages.
Wrote tarball sdist to
/usr/local/google/home/satnam/silveroak/dist-newstyle/sdist/Cava2HDL-0.1.0.0.tar.gz
Wrote tarball sdist to
/usr/local/google/home/satnam/silveroak/dist-newstyle/sdist/cava-examples-0.1.0.0.tar.gz
cabal: Error: Could not find module: NandLUT with any suffix:
["gc","chs","hsc","x","y","ly","cpphs","hs","lhs","hsig","lhsig"]. If the
module is autogenerated it should be added to 'autogen-modules'.
```

blaxill, 2021-05-17:
`cabal repl` is the cabal equivalent for getting `ghci` up, that is working for me.

blaxill, 2021-05-17:
You should never need to install any haskell packages with the cabal setup, its automatic