project-oak/silveroak #817: Global constants in firmware

jadephilipoom, 2021-06-07:
Right now the firmware verification has a lot of kludges to deal with global constants, and it's a major pain point (for example, [here](https://github.com/project-oak/silveroak/pull/815#discussion_r646733217)). I'd like to see if we can do better.

 The constraints here are:
1) protect against a global constant being overwritten with a local value (so `GLOBAL = x; f(GLOBAL)` should call `f(x)`, not `f(<literal value of the constant>)`)
2) produce C code that uses the global constant values, not arguments/literals 

I'm not sure how to resolve these constraints in a nice way. The solution we currently have is pretty roundabout and hacky (a typeclass that states what constants exist and instances of the typeclass that give string, variable, literal, Z, and word values of the constants for different contexts; string/variable in the implementations, word in the proofs, and literals for the compiler). 

Ideally, I think I would want to write the functions so they didn't take any extra arguments (although I'd be okay with a list specifying what global values they require) and reference the globals as variables. Then in the proofs, I'd want to start with a locals map that had the global values initialized as part of the locals map (again, would be okay with needing to write out their values). For the compiler I'd want to be able to specify particular Z values, or store literals somewhere in memory in the `main` function.

@samuelgruetter , do you think there are potential upstream changes that  we could make to improve this situation? For instance, could we make a version of `WeakestPrecondition.call` that would takes an initial map of global values to start with?

CC @dayeol

satnam6502, 2021-06-07:
Noted. This is too technical for me to have an opinion on what should be
done or how, but it does seem like a pain-point
worth addressing/prioritizing.

[image: Google Logo]
Satnam Singh
Software Engineer, Google Research
***@***.***

samuelgruetter, 2021-06-08:
I'm trying out an approach where the source code section in `Aes.v` is parameterized over a `Context {constant_vars : aes_constants expr}`, whose fields are then instantiated to `expr.literal SOME_Z_CONST` for proofs and for compilation using the bedrock2 compiler, and instantiated to `expr.var STRING_NAME_OF_CONST` for pretty-printing to C code, and the pretty printer gets a list of varnames that it should treat as globals.
There might be a PR soon, depending on how many more word flag shift sideconditions etc need updating... :wink:

PR #818 (2021-06-15): Don't pass globals as arguments to bedrock2 functions
Here's my attempt to fix #817.

The idea is to parameterize the source code section in `Aes.v` over a `Context {constant_vars : aes_constants expr}`, whose fields are then instantiated to `expr.literal SOME_Z_CONST` for proofs and for compilation using the bedrock2 compiler, and instantiated to `expr.var STRING_NAME_OF_CONST` for pretty-printing to C code, and the pretty printer gets a list of varnames that it should treat as globals, that is, it does not make local definitions for them, and it checks that they are not modified or used as a [... cut]
closed 2021-06-15 by jadephilipoom