project-oak/silveroak #877: Cava2 type embedding

blaxill, 2021-08-09:
Currently we are experimenting with the the following types:

```
  Inductive type :=
  | Unit : type                                                                                                                                                                                                                               | Bit  : type
  | Vec  : type -> nat -> type                                                                                                                                                                                                                | Pair : type -> type -> type
  .
```

Where `Vec Bit _` gets denoted as `N` and other `Vec T _` denote to a `list (denote T)`, perhaps we should simplify things here. 

Some thoughts from Jade here:
https://github.com/project-oak/silveroak/pull/873#discussion_r685003640_

jadephilipoom, 2021-08-09:
To make my suggestion from the comment a little more concrete, I'm suggesting something like:
```coq
Inductive type :=
| Unit : type
| BitVec : nat -> type
| Vec : type -> nat -> type
| Pair : type -> type -> type
```
`BitVec` gets denoted as `N` and only numeric operations are defined on it (no `append`; you'd have to do a shift and or).

jadephilipoom, 2021-08-10:
I went to test out some proofs, and I found that the current construction is quite difficult to reason about, especially if the types are not completely concrete. For example, let's say I want to prove this lemma about `vec_as_tuple`, which essentially says that `vec_as_tuple (x0 :: xs) = (x0, vec_as_tuple xs)`:
```coq
Lemma step_vec_as_tuple_cons {t n m} x0 xs  :
  step (vec_as_tuple (t:=t) (n:=S n)) tt (x0 :: xs, tt)
  = (tt, (x0, snd (step (vec_as_tuple (t:=t) (n:=n)) tt (xs, tt)))).
```
In fact, this lemma doesn't typecheck, because if `t` is `Bit`, then the input type is not in fact a list. I can restrict t to a `Vec` type to make the statement typecheck, even though that makes the lemma slightly less useful:
```coq
Lemma step_vec_as_tuple_cons {t n m} x0 (xs : list (denote_type (Vec t m))) :
  step (vec_as_tuple (t:=Vec t m) (n:=S n)) tt (x0 :: xs, tt)
  = (tt, (x0, snd (step (vec_as_tuple (t:=Vec t m) (n:=n)) tt (xs, tt)))).
```
However, actually _proving_ this lemma is difficult, because, internally, `denote_to_simple_denote` only half-simplifies because it doesn't know what `t` is, leaving ugly goals like:
```coq
  (tt,
  simple_denote_to_denote
    (hd simple_default
       (match t as t1 return ((denote_type t1 -> simple_denote_type t1) -> denote_type (Vec t1 m) -> simple_denote_type (Vec t1 m)) with
        | [] => fun (f : denote_type [] -> simple_denote_type []) (x : denote_type (Vec [] m)) => map f x
        | Bit => fun (_ : denote_type Bit -> simple_denote_type Bit) (x : denote_type (BitVec m)) => Vector.to_list (N2Bv_sized m x)
        | Vec t0 n0 =>
            fun (f : denote_type (Vec t0 n0) -> simple_denote_type (Vec t0 n0)) (x : denote_type (Vec (Vec t0 n0) m)) => map f x
        | t0 ** t1 => fun (f : denote_type (t0 ** t1) -> simple_denote_type (t0 ** t1)) (x : denote_type (Vec (t0 ** t1) m)) => map f x
        end denote_to_simple_denote x0
        :: map
             (match
                t as t1 return ((denote_type t1 -> simple_denote_type t1) -> denote_type (Vec t1 m) -> simple_denote_type (Vec t1 m))
              with
              | [] => fun (f : denote_type [] -> simple_denote_type []) (x : denote_type (Vec [] m)) => map f x
              | Bit =>
                  fun (_ : denote_type Bit -> simple_denote_type Bit) (x : denote_type (BitVec m)) => Vector.to_list (N2Bv_sized m x)
              | Vec t0 n0 =>
                  fun (f : denote_type (Vec t0 n0) -> simple_denote_type (Vec t0 n0)) (x : denote_type (Vec (Vec t0 n0) m)) => map f x
              | t0 ** t1 =>
                  fun (f : denote_type (t0 ** t1) -> simple_denote_type (t0 ** t1)) (x : denote_type (Vec (t0 ** t1) m)) => map f x
              end denote_to_simple_denote) xs),
    vector_as_tuple n (Vec t m)
      (tl
         (match
            t as t1 return ((denote_type t1 -> simple_denote_type t1) -> denote_type (Vec t1 m) -> simple_denote_type (Vec t1 m))
          with
          | [] => fun (f : denote_type [] -> simple_denote_type []) (x : denote_type (Vec [] m)) => map f x
          | Bit => fun (_ : denote_type Bit -> simple_denote_type Bit) (x : denote_type (BitVec m)) => Vector.to_list (N2Bv_sized m x)
          | Vec t0 n0 =>
              fun (f : denote_type (Vec t0 n0) -> simple_denote_type (Vec t0 n0)) (x : denote_type (Vec (Vec t0 n0) m)) => map f x
          | t0 ** t1 =>
              fun (f : denote_type (t0 ** t1) -> simple_denote_type (t0 ** t1)) (x : denote_type (Vec (t0 ** t1) m)) => map f x
          end denote_to_simple_denote x0
          :: map
               (match
                  t as t1 return ((denote_type t1 -> simple_denote_type t1) -> denote_type (Vec t1 m) -> simple_denote_type (Vec t1 m))
                with
                | [] => fun (f : denote_type [] -> simple_denote_type []) (x : denote_type (Vec [] m)) => map f x
                | Bit =>
                    fun (_ : denote_type Bit -> simple_denote_type Bit) (x : denote_type (BitVec m)) => Vector.to_list (N2Bv_sized m x)
                | Vec t0 n0 =>
                    fun (f : denote_type (Vec t0 n0) -> simple_denote_type (Vec t0 n0)) (x : denote_type (Vec (Vec t0 n0) m)) => map f x
                | t0 ** t1 =>
                    fun (f : denote_type (t0 ** t1) -> simple_denote_type (t0 ** t1)) (x : denote_type (Vec (t0 ** t1) m)) => map f x
                end denote_to_simple_denote) xs)))) =
  (tt,
  (x0,
  simple_denote_to_denote
    (vector_as_tuple n (Vec t m)
       (map
          (match
             t as t1 return ((denote_type t1 -> simple_denote_type t1) -> denote_type (Vec t1 m) -> simple_denote_type (Vec t1 m))
           with
           | [] => fun (f : denote_type [] -> simple_denote_type []) (x : denote_type (Vec [] m)) => map f x
           | Bit => fun (_ : denote_type Bit -> simple_denote_type Bit) (x : denote_type (BitVec m)) => Vector.to_list (N2Bv_sized m x)
           | Vec t0 n0 =>
               fun (f : denote_type (Vec t0 n0) -> simple_denote_type (Vec t0 n0)) (x : denote_type (Vec (Vec t0 n0) m)) => map f x
           | t0 ** t1 =>
               fun (f : denote_type (t0 ** t1) -> simple_denote_type (t0 ** t1)) (x : denote_type (Vec (t0 ** t1) m)) => map f x
           end denote_to_simple_denote) xs))))
```
To resolve this, I think we should either have a primitive `BitVec` or treat _all_ `Vec`s, including `Vec Bit`s, as lists/vectors.

blaxill, 2021-08-10:
I'm leaning towards primitive `BitVec`s, but before I implement it, does anyone prefer the all Vec=list representation? @jadephilipoom

samuelgruetter, 2021-08-10:
> current construction is quite difficult to reason about, especially if the types are not completely concrete

I also got the impression that if the type parameters passed to `Circuit` are not completely concrete, it's difficult to get things to typecheck.
In @jadephilipoom's example above, it can probably be fixed by introducing a primitive `BitVec` type and simplifying `denote_type`, so that `denote_type` is convertible as expected, but I think there will also be cases where this is not enough. For instance, `absorb_any` first matches on its first argument, and then on its second argument, so if you have a `[] ++ x`, you're lucky because it is convertible to `x`, but if you have a `x ++ []`, you're out of luck, because it's not convertible to `x`. I reported such an example at https://github.com/project-oak/silveroak/issues/884.

PR #885 (2021-08-11): Uncouple BitVec and Vec
Resolves #877 

- Makes `BitVec` a constructor of the `type` inductive, replacing `Bit`
- Removes `simple_denote_type`; it's no longer necessary if `BitVec` is always `N` and `Vec` is always `list`
- Removes `take` and `drop` in favor of `skipn` and `firstn` (they appeared to be exactly equivalent; any reason to use our own custom definitions instead of the standard library here?)
- Adjusts primitives so that "arithmetic"-type operations (e.g. shifting, addition, bitwise operations) are defined only on `BitVec`, and "vector"-type operations (e.g. uncons, reverse) are de [... cut]
closed 2021-08-11 by jadephilipoom