project-oak/silveroak #893: Missing directory dependencies on CI

samuelgruetter, 2021-08-12:
The end-to-end theorem that I'd like to sketch by the end of next week will introduce the following three dependencies which are currently not taken into account by the CI (where `<--` means "depends on"):

* `firmware/Hmac/HmacProperties.v` `<--` `silveroak-opentitan/hmac/Spec/SHA256.v`
* `investigations/cava2/Sha256Properties.v` `<--` `silveroak-opentitan/hmac/Spec/SHA256.v`
* A file with an end-to-end theorem about Sha256, either living in `firmware/Hmac` (similarly to `firmware/IncrementWait/IncrementWaitToCava.v`), or in a separate directory `<--` `investigations/cava2/`

To start, I can of course just build the right subdirectories locally & manually, but it would be nice if this was also supported on CI.

jadephilipoom, 2021-08-17:
I suggest:
- Move `investigations/cava2` out of `investigations` and make it a top-level directory that is run by CI
- Move the HMAC-specific circuit implementations out of `cava2` and move it into a new directory `silveroak-opentitan/hmac/hw` (since we eventually want `cava2` to be just the core library, it's probably good not to have a bunch of specific code there)
- Move the HMAC-specific firmware out of `firmware` and into a new directory `silveroak-opentitan/hmac/sw` (less sure of this one, but it might make sense to organize things by device for bigger projects like HMAC -- an unfortunate side effect is that the `run_silicon_creator_tests_in_opentitan` script will have to be copied)
- Move the end-to-end proofs into a new directory `silveroak-opentitan/hmac/end2end`
- Introduce the necessary CI dependencies:
  -  `silveroak-opentitan/hmac` `<--` `cava2`
  -  `silveroak-opentitan/hmac` `<--` `firmware`
- Remaining dependencies will be handled in `silveroak-opentitan/hmac/Makefile`:
  -  `silveroak-opentitan/hmac/hw` `<--` `silveroak-opentitan/hmac/Spec`
  -  `silveroak-opentitan/hmac/sw` `<--` `silveroak-opentitan/hmac/Spec`
  -  `silveroak-opentitan/hmac/end2end` `<--` `silveroak-opentitan/hmac/hw`
  -  `silveroak-opentitan/hmac/end2end` `<--` `silveroak-opentitan/hmac/sw`

We could also anticipate future slowness and separate the four `silveroak-opentitan/hmac` directories into their own CI jobs so that e.g. `hw` and `sw` can run in parallel.

jadephilipoom, 2021-08-17:
Adding a note we talked about verbally during the project meeting: for #904, we should also preemptively introduce the CI dependency that `firmware` depends on `cava2`.

jadephilipoom, 2021-08-19:
I think #910 resolved this one. We still might want to separate out the `Util` directory to further improve our dependency structure, but I think that's a separate improvement.

closed 2021-08-19 by jadephilipoom